betaflight/betaflight #3788: beeper -DISARM_REPEAT disables RX_SET beeper when GPS is enabled (and working)

bkeinert, 2017-08-08:
It seems that "beeper -DISARM_REPEAT" disables the remote controllable beeper mode when GPS is enabled (+3D fix and >= 5 satellites).

Reproduce:
- enable feature GPS (with an actual GPS connected...)
- setup an AUX channel controlled beeper
- beeper -DISARM_REPEAT
- wait for 3D GPS fix with >=5 satellites and the AUX controlled beeper doesn't work anymore

Things I noticed from a rough look over the sourcecode:
- beeper.c:320 `if (STATE(GPS_FIX) && gpsSol.numSat >= 5) [...]` this causes BEEPER_MULTI_BEEPS to be used instead of BEEPER_RX_SET.
- the entries for DISARM_REPEAT and BEEPER_MULTI_BEEPS are next to each other in the `beeperTable`. I suspect there might be an issue where DISARM_REPEAT effectively disables BEEPER_MULTI_BEEPS (off by one somewhere? Maybe something around beeper.c:376?).

bkeinert, 2017-08-12:
Looked into the code again and I think the issue might be caused by the enum `beeperMode_e`  (beeper.h:23) being not "in sync" with `beeperTable` (beeper.c:190).
I think `BEEPER_DISARM_REPEAT` should just be moved below `BEEPER_MULTI_BEEPS` in the enum. This should fix this and probably some minor other issues with the beeper flag mask.

Seemingly the CLI code uses `beeperNameForTableIndex(i)` and `mask = 1 << i;`, while other parts of the code use values from the enum (minus one). I guess either the table and the enum have to be in sync or the CLI code should call `beeperModeForTableIndex(i)` to retrieve the correct value for the mode.

Yep, I know this isn't a super important issue, but maybe still worth fixing (and probably super easy to fix).

mikeller, 2017-08-13:
Good find, @bkeinert, thanks for looking into this!

Do you want to submit a pull request to fix this (this should definitely go into 3.2)?

bkeinert, 2017-08-13:
I can do that (it will take a short moment to figure out how github works, but that isn't a problem).
For which variant of fixing the issue should I go? The easy enum fix or (<- logical or) should I also touch the CLI code?

mikeller, 2017-08-13:
No problem, take your time, and feel free to ask questions if you need help (either in here, or join the Betaflight slack on http://www.betaflight.ch/).

I think the correct solution is to fix this in the CLI. The enum order matters for priority, so changing it will change the behaviour, and is probably best avoided.

bkeinert, 2017-08-13:
I think probably both should be done, since `beeperTable` also contains a priority which currently doesn't match the enum. Didn't look yet into which one is really used for actual behavior - if both ways are used throughout the code, they should probably match.

closed 2017-08-14 by bkeinert