betaflight/betaflight #3791: Matek f405 AIO w/ BF 3.2 won't arm or read battery volatge

DeltaV1, 2017-08-09:
When I plug in the battery the ESCs make the 3 tone beeps but, don't make the following 2 beeps which I assume means that they aren't getting a signal from the FC and therefore won't arm. I know it's not a problem with the ESCs because they worked just fine with a different FC that was also running 3.2 and dshot 600.

Also, when the battery is plugged in the battery voltage isn't displayed on either the OSD or in the configurator. 

I'm also having this issue where at the bottom of the configurator the cpu load, cycle time, and I2C error are all reading nothing except zero.

jflyper, 2017-08-09:
@DeltaV1 I think you have two separate issues:

(1) Non-fully erased FC. Try full-chip erase or cli "defaults".

(2) Your BFC is out of sync with BF.

DeltaV1, 2017-08-09:
@jflyper 

(1) The fc has already been fully erased.

(2) If it is out of sync how could I get it to?

Saxin, 2017-08-09:
@DeltaV1 
I have such board with 3.2RC2 - works flawlessly, flies @ 32/16 (OC is ON).

the-scriptinator, 2017-08-10:
I have a copy of the non AIO matek board and have the same issue.
After installing the BF3.2 release candidate and turning on warnings in the OSD I get a "No Gyro" message.
Sometimes the FC eventually fully boots but it can take up to 2-5 minutes - I've timed it.
Pulling he battery and re-connecting, sometimes a number of times seems to help.

This FC has had this issue from the start and I also had another copy that had same issue.

I also have copies of the Matek AIO and Matek Non AIO that work fine.

I suspect quality issues.

robert-hoffmann, 2017-08-16:
I have the matek f405 aio (not copy) running RC3.
Currently no issues to report.

AndyCWilson, 2017-08-16:
I have the matek f405 AIO and it works great. No issues here either.

jflyper, 2017-08-18:
I'm closing this.

@DeltaV1 If you still have the issue, please redirect it to RCG thread:
https://www.rcgroups.com/forums/showthread.php?2889298-MATEKSYS-Flight-Controller-F405-32K-Gyro-BFOSD-5xUARTs-SD-Slot

closed 2017-08-18 by jflyper