betaflight/betaflight #3792: Kombini FC only finding 3 ESCs with 3.2.0

ianpaine, 2017-08-09:
Upgraded to 3.1 but was greeted with 5 beeps on power up and only 3 ESCs found/motors working in motor tab. Revert back to 3.1.7 and everything is working as it should.

claytonbakerjr, 2017-08-09:
Curious...is it motor 1 that is not working? I ended up using the PICOBLX firmware and that resolved my problem with my Radiance board however I believe there is a CLI command you can type in to possibly fix your current problem. I unfortunately am not in the position to find it now but I am sure someone will pop in.

ianpaine, 2017-08-09:
I will have to re-flash and check which it was, I was unfortunately in a bit of a panic to get it working for the weekend and only just got around to reporting.

jflyper, 2017-08-09:
@ianpaine This is what you want: #3762.

@MJ666 I think FF needs to have entries for motor 1 woes in 3.2, in each corresponding board wiki.

wolverin3, 2017-08-10:
Actually, using BF3.2 RC3... I flash it to my Kombini non-DShot and enable DShot, it is working right away.

The only thing I did earlier was to bridge Motor 1 signal pin to PPM pin. That is all.

I didn't do any resource mapping in CLI for this to work.

wolverin3, 2017-08-10:
I'm using the FF_Kombini hex.

ianpaine, 2017-08-10:
I'm not running DSHOT, just multishot

emmanuelmota, 2017-08-15:
I'm having the same issue. Using betaflight_3.2.0_FF_KOMBINI-76-53920103d.hex.
T-motor DSHOT escs that I run using multishot on 3.1.7  
BL Heli only sees 3 ESCs.

mikeller, 2017-08-16:
Duplicate of #3762.

closed 2017-08-16 by mikeller

emmanuelmota, 2017-08-16:
Thank you Michael!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>