betaflight/betaflight #3793: Blackbox save onboard log freeze download box

TomasKot, 2017-08-09:
I have Naze rev6 with 3.2.0 RC3 and I can't download logs from flights. I can see it is saved on the onboard chip, but when I want to save it to my disk then the save popup freezes and no progress of savings. Unable to click on cancel, I can still see the status bar CPU works but that's all. Just working unplugging the FC.

atomiclama, 2017-08-11:
Confirmed on CC3D with a personal build from latest master on 10/8 will get the commit in a bit.

atomiclama, 2017-08-12:
@DieHertz what do you think?

DieHertz, 2017-08-12:
@atomiclana no idea, Huffman isn't enabled for F1 AFAIK, maybe some collateral damage, I'll have a look tonight or tomorrow.

djmartin, 2017-08-14:
I'm also having the same issue downloading blackbox logs on SPRacing F3 from onboard flash.  Worked fined on RC2 - broken in RC3

DieHertz, 2017-08-14:
Weird, it works for me on a BlueJay F4 with builtin SPI flash and on a XRacer F303 (SPRacing F3 target).

djmartin, 2017-08-14:
Could be a combination of other settings?
Log file seems to be on the flash, just doesn't save. (0KB file created).
Would any log files help diagnose... please let me know what you need.

DieHertz, 2017-08-14:
I'll have another look tonight, just give me your diff and check chrome console for errors

djmartin, 2017-08-14:
Here is the CLI diff.  I'm out of time to run the test in the console, but will have another try later.

Hope this helps

[BTFL_cli_DavidGT90_20170814_200619.txt](https://github.com/betaflight/betaflight/files/1221881/BTFL_cli_DavidGT90_20170814_200619.txt)

djmartin, 2017-08-14:
I can also try some more testing on 3x other quads (ImpusleF3, FuryF4 & ColibriRace), but won't be until I get time again next weekend

DieHertz, 2017-08-15:
I have tried downloading blackboxes with latest master of BF and BFC and it works on both SPRF3 and BJF4 for me.
But there's still the Chrome-related (?) issue when most of our jumbo-frame packets fail the CRC check. To support this theory, please connect to your FC, open Chrome Developer Console for BFC main window, type `TABS.onboard_logging.BLOCK_SIZE = 512` and try downloading your log again.

djmartin, 2017-08-16:
Thanks Andrey.  I think you are correct in this is the cause of the issue I'm experiencing. How do I get to the Chrome Developer Console when I'm in BetaFlight Configurator?  I usually just hit CTRL+SHIFT+J in Chrome.
Happy to test this out, but I'm certainly a novice in this area.

DieHertz, 2017-08-16:
If `C-S-j` works for you then that's it, open console that way and try changing block size.

djmartin, 2017-08-16:
Unfortunately c-s-j doesn't work when I fire up BetaFlight Configurator... only in my normal Chrome window... is there a "trick"?

DieHertz, 2017-08-16:
You probably have to enable developer mode on the chrome://extensions tab, or use "inspect main.html" from there directly when BFC is running

djmartin, 2017-08-17:
OK... still not working with block size = 512.
Here is my Console when I try to export:
Libraries: jQuery - 2.1.4, d3 - 3.5.6, three.js - 72
port_handler.js:71 Selecting last used port: COM4
serial_backend.js:49 Connecting to: COM4
serial.js:134 SERIAL: Connection opened with ID: 2, Baud: 115200
serial.js:55 {connectionId: 2, error: "overrun"}
watch_for_on_receive_errors @ serial.js:55
EventImpl.dispatchToListener @ extensions::event_bindings:380
publicClassPrototype.(anonymous function) @ extensions::utils:149
EventImpl.dispatch_ @ extensions::event_bindings:364
dispatchArgs @ extensions::event_bindings:237
dispatchEvent @ extensions::event_bindings:246
serial.js:108 SERIAL: Connection recovered from overrun condition
msp.js:210 MSP data request timed-out: 10
three.min.js:536 THREE.WebGLRenderer 72
TABS.onboard_logging.BLOCK_SIZE = 512
512
onboard_logging.js:450 Dataflash dump file path: C:\3.2.0\Blackbox\BTFL_BLACKBOX_LOG_DavidGT90_20170817_200914.BBL
onboard_logging.js:344 Received 0 bytes in 0.08s (0.00kB / s) with block size 512.

djmartin, 2017-08-17:
[image]

djmartin, 2017-08-17:
Saves a 0KB file

DieHertz, 2017-08-17:
The `BLOCK_SIZE` was meant for those who experience "Freezing", in your case there's no freeze - it just downloads no data and finishes successfully.

djmartin, 2017-08-20:
So confirming I still have this issue of saving a 0KB file.  
As per screen shot, BlackBox is storing space of the Onboard Flash, just not saving.
Not sure if this is an issue with Betaflight Code or Configurator.  Was working last week when I had 3.2.0 RC2 on this same quad - but perhaps Configurator also was updated?
Any suggestions on what I can try to help to debug this further?

djmartin, 2017-08-20:
Also confirming I only have this issue on 1 of my Quads - ImpulseF3 + Fury F4 are both working fine.. only on my Micro KingKong GT90 SPRF3

DieHertz, 2017-08-20:
If you can use configurator from GitHub, I could give you a small patch which comments one line to disable compression, to see if it's related to your issue

djmartin, 2017-08-20:
OK... can give it a try.  Any reason why this issue would only happen on my GT90 & not my other quads?
Please let me know what patch I need to download to test this.
Thanks

DieHertz, 2017-08-20:
https://github.com/DieHertz/betaflight-configurator/tree/disable-blackbox-compression
pull this branch from my repository

djmartin, 2017-08-20:
Thanks.  WIll give it a try when I get some spare time tomorrow.
Been a long time since I did this :)

djmartin, 2017-08-20:
OK... just realised you've done all the hard work, so I just fired it up & tested it then.
Yes, confirming this version works & saves the file correctly :)

djmartin, 2017-08-20:
Here is the console log information if that helps
Libraries: jQuery - 2.1.4, d3 - 3.5.6, three.js - 72
port_handler.js:77 Last used port wasn't saved "yet", auto-select disabled.
port_handler.js:99 PortHandler - Found: COM4
serial_backend.js:49 Connecting to: COM4
serial.js:134 SERIAL: Connection opened with ID: 3, Baud: 115200
three.min.js:536 THREE.WebGLRenderer 72
19main.html#:1 Error in event handler for serial.onReceive: TypeError: Cannot read property 'toString' of null
    at chrome-extension://binimcbkfigiffooafagfefnnkojffmp/tabs/setup.js:174:157
    at MspHelper.process_data (chrome-extension://binimcbkfigiffooafagfefnnkojffmp/js/msp/MSPHelper.js:1176:31)
    at chrome-extension://binimcbkfigiffooafagfefnnkojffmp/js/msp.js:139:13
    at Object.notify (chrome-extension://binimcbkfigiffooafagfefnnkojffmp/js/msp.js:138:24)
    at Object.read (chrome-extension://binimcbkfigiffooafagfefnnkojffmp/js/msp.js:126:26)
    at read_serial (chrome-extension://binimcbkfigiffooafagfefnnkojffmp/js/serial_backend.js:362:13)
msp.js:210 MSP data request timed-out: 101

DieHertz, 2017-08-20:
I'm confused by this `toString`invocation on `null`, and it puzzles and disappoints me that compression doesn't work for you, but I'll try to solve it. Unfortunately I can't reproduce it yet on my own SPRF3.

djmartin, 2017-08-23:
Please let me know if there is anything else I can do to help solve / test this issue.
I'll just keep using your modified version to download blackbox logs on this quad for now.  :)

riquezjp, 2017-09-07:
Just adding that I have this problem on DYS F4. BF3.2 RC4, I cannot download logs using configurator 3.2.1 
The progress bar moves exceptionally slowly, the logs created are around 500k after many hours. Even left overnight it does not complete more than 10%

Using DieHertz fork I was able to download logs successfully / normally in 20 mins or so.

atomiclama, 2017-09-08:
Is this a duplicate of #4086 ? Fixed by #4088 
Gents @TomasKot @riquezjp @djmartin  can you check ?

I don't have access to a F1 board at the mo.

DieHertz, 2017-09-08:
This indeed is a duplicate, I just realized ticket starter has a Naze. My code had a bug which made F1 boards return no bytes if configurator requested compression.

djmartin, 2017-09-09:
Hi Andrey, confirming that when I flashed 3.2.0 RC4 this fixed the issue for me.  So RC2 + RC4 both working - RC3 no good - except via your special debug version of Configurator. :)
Hope that helps.

DieHertz, 2017-09-09:
I think we're good on this one then

djmartin, 2017-09-09:
Agree :)

mikeller, 2017-09-11:
@TomasKot: Has this been solved for you?

closed 2018-04-21 by DieHertz