betaflight/betaflight #3802: [Feature Request] Disable stats screen, show voltage

riquezjp, 2017-08-10:
Often I fly around, land, change some PID or whatever & I would like to see the battery voltage after disarming to decide if I should take another flight.
Currently I need to arm again, or enter OSD & save, to see that.

This may be accomplished by having an option to disable the stats screen, or by having a 'current voltage' on the stats instead of 'end voltage' which i find to be quite inaccurate.

Thanks for your consideration.

elektronisk, 2017-08-10:
Maybe see this in context with #3644 about multiple osd pages. It was suggested there to have the stats screen replaced by just another osd page (that is activated when disarmed). End voltage, delta mAh, max current etc. should in this proposal be made into general osd elements. This way people can choose what they want to see.

DanNixon, 2017-08-10:
You can dismiss the stats screen by moving either the pitch or throttle stick high, this will bright you back to the "main" OSD screen (which I assume you have a battery voltage indicator on).

riquezjp, 2017-08-10:
oh really? I didnt know that. Thanks Dan!

closed 2017-08-12 by mikeller