betaflight/betaflight #3804: DYS F4 3.2 not reading battery voltage

N1tor, 2017-08-10:
I have tried all of the inputs in the battery tab, but BF still shows the voltage at 0 in both the OSD and chrome app

App version: 3.2.1
Firmware: Omnibuss F4 3.2 RC3

Currently running my DYS F4 in 3.16, where the voltage works fine in both the OSD and in the chrome app

AndersHoglund, 2017-08-10:
Try target DYSF4PRO

N1tor, 2017-08-10:
Did not see that there was a new target for the DYS board!

But sadly the issue is in this version as well

jflyper, 2017-08-10:
@N1tor The voltage and current meters are working with DYSF4PRO target without any problem.

Do a full erase or cli "defaults", immediately goto battery tab and select Onboard ADC for voltage and current sources, hit save and power cycle.

AndersHoglund, 2017-08-10:
This seems like a duplicate of issue #3770 . 
Do you guys also have the DYS F20A/F30A 4-in-1 ESC, or similar ? Battery connected to ESC? Battery must be connect via FC to ESC for current measurement to work. Vbat should be connect via the pin header and work OK that way. But if you have some other connections FC - ESC, please check that vbat is properly connected.  Can not do much about this in the firmware.

N1tor, 2017-08-10:
Had to reboot the board a couple of times and got it working! Thank you

closed 2017-08-10 by N1tor