betaflight/betaflight #3806: Motor #3 slower than others

TomTiki, 2017-08-10:
Hi Boris and everyone in the Betaflight community,

I'm having a problem with my Eachine Wizard that is driving me nuts.  Everything had been working fine and then suddenly i found that motor 3 (rear left) started lagging and barely has any torque.  The result is that the quad almost flips when trying to takeoff, drifts back and left in a hover and has a severe wobble on a punch-out.  The quad has an SP Racing F3 flight controller and I am running Betaflight 3.1.7.  Here is the troubleshooting I have done:

-Replaced motor #3.  No improvement.
-Swapped ESC #3 with #1.  No improvement.
-Recalibrated all ESCs.  No improvement.
-Installed new FC.  No improvement.

jflyper, 2017-08-10:
@TomTiki We are really sorry to tell you that the Betaflight github issues is not a place for user support, but is a place for reporting software issues and discussion toward making the Betaflight software better.
Please redirect your support request to appropriate places on the net, such as the dedicated thread for your quad. Thank you in advance.

Ref: (Is this your quad?)
https://www.rcgroups.com/forums/showthread.php?2892388-Eachine-Wizard-X220S-ARF-RTF-Upgraded-Wizard-w-F4-BF-OSD-DShot-600-30A-ESCs

TomTiki, 2017-08-10:
ok thanks @jflyper.   I'll redirect my request elsewhere

closed 2017-08-11 by jflyper