betaflight/betaflight #3812: TIM1_CH2N and TIM1_CH3N are not edge interrupting

jflyper, 2017-08-11:
Some targets use `TIM1_CH2N` on PB0 and `TIM1_CH3N` on PB1 for software serial. Which turned out to be not operational. Investigation revealed that these timer channels are not interrupting on edge , while overflow is interrupting, means it can transmit but can not receive.

Related with #3549?

The investigation was carried out using OMNIBUSF4SD target with timer mapping that forces PB0 and PB1 to use `TIM1_CH2N` and `TIM1_CH3N`:
```
#if 1
    // Testing softserial over n-channel (doesn't work)
    DEF_TIM(TIM1,  CH2N,PB0,  TIM_USE_NONE,                TIMER_OUTPUT_INVERTED, 0),
    DEF_TIM(TIM1,  CH3N,PB1,  TIM_USE_NONE,                TIMER_OUTPUT_INVERTED, 0),
#else
    // This is the original mapping (works)
    DEF_TIM(TIM3,  CH3, PB0,  TIM_USE_MOTOR,               TIMER_OUTPUT_STANDARD, 0), // S1_OUT D1_ST7
    DEF_TIM(TIM3,  CH4, PB1,  TIM_USE_MOTOR,               TIMER_OUTPUT_STANDARD, 0), // S2_OUT D1_ST2
#endif
```

jflyper, 2017-08-11:
TIM1 CC is not interrupting at all.
- Hooked `timCCxHandler`: TIM3 interrupt was hooked, TIM1 nothing.
- `DIER` value: `TIM3 = 10001b = (CC4IE | UIE)`, `TIM1 = 01001b = (CC3IE | UIE)`, which is correct.

jflyper, 2017-08-11:
TIM1 CC is not interrupting (consistently) on non-N-channels, too.

However, there is a tiny difference in behavior: it interrupts for 3 times and quits.

Used backdoor timers of UART1 to check this out.

```
    DEF_TIM(TIM1,  CH2, PA9,  TIM_USE_NONE,                TIMER_OUTPUT_NONE,     0), // UART1_TX
    DEF_TIM(TIM1,  CH3, PA10, TIM_USE_NONE,                TIMER_OUTPUT_NONE,     0), // UART1_RX
```

jflyper, 2017-08-11:
@ledvinap Any insight into this problem? I'm totally stumped.

jflyper, 2017-08-11:
TIM8 works, so this is not related with advanced timers...

```
    DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_PWM,                 TIMER_OUTPUT_NONE,     0), // S5_IN
    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_PWM,                 TIMER_OUTPUT_NONE,     0), // S6_IN
```

ledvinap, 2017-08-18:
@jflyper : this is strange ... using N channel should not make any difference at all. And handler code is identical for all timers (and is used for PPM), so it should affect more than softserial ...
Can you verify  PPM/PWM receiver on TIM1?

Maybe something in softserial is hardcoded, but even that is strange ...

jflyper, 2017-08-21:
@ledvinap PPM doesn't work on TIM1 CH2N
```
    DEF_TIM(TIM1,  CH2N,PB0,  TIM_USE_PPM,                 TIMER_OUTPUT_INVERTED, 1), // (2,2)
```

jflyper, 2017-08-21:
@ledvinap There's something going on with these channels. Did you see #3549?

ledvinap, 2017-08-29:
Sorry, almost forgot about this one ... and studied TIMs a bit:

N channel can't be used as input - only TIMx_CH[1-4] are routed to input (and some multiplexing can be done between pair of channels, it's implemented in timer code)

It will be necessary to use TIM3_CH[34] for softserial ...

jflyper, 2017-08-29:
@ledvinap Yeah, I did some investigation myself (including posting a question at community.st.com) and have reached the same conclusion this afternoon. I really appreciate your confirmation.

DieHertz, 2017-08-29:
Thanks @ledvinap, we have discovered this recently while studying diagrams in reference manuals for STM32 :-)

closed 2017-08-29 by jflyper

ledvinap, 2017-08-29:
In mid-term, it should be possible to configure TIM3 when input function is required, but still supporting output on TIM1