betaflight/betaflight #3818: 3.2 Jeti telemety not working in F4 OMNIBUSF4

flyingbuu, 2017-08-12:
Updated to 3.2 and jeti telemetry is not working also the serial rx ony work on uart 1 not 6 and is in the wiki

Is there workaround or quick fix?

Soureaver666, 2017-11-25:
I also tried 3.2.0 to 3.2.2 with the Jeti EX BUS protocol and it does not work anymore. It seems like it "forgets" the "JETIEXBUS" when trying to save it - other than in older versions.

What was changed in the serial receiver implementation and is there a workaround?

I don't want to choose between that new awesome dynamic notch filter and the ex bus telemetry :(

mikeller, 2017-11-25:
@Soureaver666: What flight controller are you using?

Soureaver666, 2017-11-25:
EMAX F3 Femto, it is based on the SPRACING F3EVO Firmware
https://www.banggood.com/en/Emax-STM32F303-F3-Femto-Flight-Controller-with-Integrated-BECBuzzer-PadsVBatPDB-p-1101875.html?cur_warehouse=CN

mikeller, 2017-11-25:
JETIEXBUS support has had to be removed from this target in 3.2, in order to make the firmware fit into the limited amount of flash space available. Sorry.

closed 2017-11-25 by mikeller