betaflight/betaflight #3819: BF 3.2: Disarmed motors spin up in configurator, armed motors spin irregular

jangrewe, 2017-08-12:
I just updated my Omnibus F4 Pro v3 to BF 3.2 RC3 (had RC2 before that), and when using Configurator 3.2.1, my motors spin up like they were armed when switching to the Motors tab.
If i _then_ arm, they go into a semi-spinning mode where each motor spins maybe 1-2 revolutions every couple of seconds, but they're not in sync.
I can't get them to completely stop unless i reboot the FC or pull the plug.
Also: when booting up the quad and arming by switch, they do that same irregular spinning, not the "smooth, low rpm" normal one.
I have my FC configured to use DSHOT600, but this also happens with DSHOT250.

This didn't happen the last time i tested it (about 1-2 weeks ago), everything was behaving as expected back then. Yes, i have been flying this quad. No, i don't do stick arming. ;-)

jangrewe, 2017-08-12:
I did a full chip erase and this behaviour stopped, but i'll leave this here just in case somebody else starts freaking out about "BF 3.2 IS ARMING BY ITSELF" ;-)

mikeller, 2017-08-13:
@jangrewe: This is a known problem, and a shortcoming of how version numbers are tracked in the firmware (the RC part is ignored there). Normally, as part of a firmware upgrade, the configuration will be reset to defaults. But since the RC versions are seen as one and the same version, this does not work when upgrading between RC versions.

closed 2017-08-13 by mikeller