betaflight/betaflight #3822: MS5611 baro calibration

andrejpodzimek, 2017-08-13:
The MS5611 barometer appears to remember pressure values for a very long time without power. For example, when I moved to a field 30 meters lower in altitude than my apartment, I got a reading of -30 meters, although the aircraft had had no battery connected for more than 40 minutes. This is somewhat inconvenient. Would it be possible to just set a zero altitude upon arming, either by reseting the sensor in some way or by simply by using the initial readings as an offset?

jflyper, 2017-08-14:
@andrejpodzimek Taking a relative zero altitude at arming can be done, but before that, I think your MS5611 has some problem in measurement. It does not remember last measurement by itself. I suspect the breather hole is blocked.

BTW, according to your claim, your MS5611 should indicate and hold 0m when you go down to -30m from where you are, not -30m, isn't it?

andrejpodzimek, 2017-08-16:
The breather hole(s) of the MS5611 are covered with foam. But I don't think they are jammed and I never put any sticky tape or glue-ish stuff on them. Basically the whole baro ([soldered](https://static.rcgroups.net/forums/attachments/6/6/9/3/8/7/a10249130-16-baro0.jpg) in a somewhat crazy way) is wrapped in a foam sheet for protection. This appeared to reduce the amount of fluctuations in flight (which may have been related to propeller noise, because they appeared to be throttle-related). (Well, for context, those were small shivers of plus/minus 1 meter of altitude, not [this disaster](https://github.com/betaflight/betaflight/issues/3861) yet.)

When I connect the battery, arm the drone and walk or fly 30 meters lower, the baro should (of course) indicate -30 meters. When I reset the aircraft completely, I'd expect an indicated altitude of 0 initially. Perhaps disarming could also reset the altitude, but that's a matter of personal preference.

mikeller, 2017-08-17:
I think it makes sense to close this one, and have the MS5611 related discussion in #3861.

andrejpodzimek, 2017-08-17:
This issue is completely unrelated to #3861, which started (together with I2C issues) after a hardware modification (compass swap). I reported this baro calibration problem back when the I2C bus was working properly and the issue will most likely be around even after/if #3861 gets fixed.

andrejpodzimek, 2017-09-01:
BTW, I just noticed that mwosd does some base altitude calibration on its own. I took off at 5 meters and landed at 3 meters and mwosd was showing -2, whereas SmartPort telemetry had the “raw” values.

martin19117, 2019-04-16:
Is there a fix for this