betaflight/betaflight #3824: Altitude hold mode functionality and documentation

andrejpodzimek, 2017-08-13:
While experimenting with the baro hold mode, I found the throttle behavior somewhat counter-intuitive. It appears that the altitude hold only kicks in when the throttle value doesn't change for a certain (short) period of time and moving the throttle stick restores the throttle to the level where the stick is at ... for a short moment. Sadly, I can't see this documented anywhere. Or it may just as well be an artifact of #3823 rather than an actual behavior. Either way, documentation on this would be helpful.

For example, when I control a Hubsan H501S (with its unknown proprietary firmware) using my Devo 10 + Deviation + 4-in-1 transmitter, switching to baro hold mode changes the meaning of the throttle signal in such a way that it determines the rate of climb / descent. -100 and 100 are the maximum rates of descent and climb, respectively, and 0 stands for a steady altitude. Then it's easy to e.g. designate a switch on the transmitter to ignore the throttle stick position and transmit 0 on the throttle channel all the time. Poor man's throttle spring loading, so to speak.

It would be great to have altitude control functionality similar to this^^^ in Betaflight. Or, at the very least, a Wiki page exactly documenting the current altitude hold mode (and the meaning of the throttle channel in that mode).

(Another interesting option would be a channel for absolute altitude, which could then be dialed in e.g. using one of the rotary potentiometers on the TX.)

andrejpodzimek, 2018-05-06:
Note to @stale: This is still a problem.

timman2er, 2018-07-28:
@andrejpodzimek I think I get what you mean, but I need clarification.

On the Hubsan, when you activate baro hold mode, the current position of ascent/descent is 0, or is the middle position 0 and the end points 100 and -100 respectfully? 

And that is how it should behave?

andrejpodzimek, 2018-07-28:
On Hubsan (and also on DJI, I believe) in baro-hold modes, middle throttle position (0) is neutral, i.e., zero vertical speed, maximum (+100) is the fastest ascent speed and minimum (-100) is the fasteste descent speed.

The position of the throttle input at the time of baro hold mode activation is never considered, because on the default TXs for DJI and Hubsan, the "throttle" (left stick vertical movement on Mode 2) is spring-loaded and kept in the center by default. Therefore it wouldn't make sense to have a neutral vertical holding position elsewhere.

I use my Devo 10 and Devo 12e (with Deviation, of course) to control my Hubsan drones. (It speaks 40+ different protocols, including Hubsan.) What I have on the Devo is a switch that can freeze the throttle channel at 0, regardless the throttle stick position, when (and only when) baro mode is active. I don't have the throttle spring-loaded on the Devo, because I use it also to fly all my crazy home-built Betaflight drones as well.

So yes, I think that's how it should behave, making '0' stand for "zero vertical speed" sounds reasonable to me, close enough to what the default TXs do and easily configurable even on a TX without spring-loaded throttle.

closed 2018-09-07 by andrejpodzimek