betaflight/betaflight #3825: Motors will spinn at power on if Tx arming switch is accidentally ON.

AndersHoglund, 2017-08-13:
Reports: on the Slack general channel copied here:

> stellarhopper
> 7:42 AM
> If my arm switch is on when plugging in, and throttle is 0, is it supposed to arm? It did today (left the switch on accidentally), and it started spinning.. No harm done, but caught me a bit by surprise.. Maybe it can bee made so that if the arming aux channel is in on position at startup, disregard it till an on-off-on transition is made

borisbstyle, 2017-08-13:
That is pretty much "normal" behaviour. No bug at all.

You can even have double or triple arming switch, but still have them all accidentally on

AndersHoglund, 2017-08-13:
Normal or not, this is a dangerous behaviour. We can NOT have it this way. There are old PRs suggesting safer solutions, and a new one comming.

AndersHoglund, 2017-08-13:
Currently it seems to be enough with a static state of arming switch being ON at initial power on. With only a simple single arm mode being configured. No Pre-Arm or anything else like that set up.

What should be required as a bare minimum is a transition from dis-armed to armed state initially. There are a number of other edge cases too, hopefully covered by the coming PR.

DanNixon, 2017-08-13:
https://github.com/betaflight/betaflight/pull/3826 is my preferred solution, this covers a few edge cases that could catch someone out if they haven't taken the time to understand how the arming sequence works.

chrvog, 2017-08-16:
@borisbstyle I was not able to reproduce this behavior on any of my quads before the update to 3.2. It's at least a changed behavior for most people and it caught me by surprise on the field.

mikeller, 2017-08-17:
Fixed by #3826.

closed 2017-08-17 by mikeller