betaflight/betaflight #3833: Omnibus F4 Pro with L9R -> RSSI output is not correct

Tiberius131, 2017-08-13:
Hey,

Im trying to get a correct rssi value in my osd.
I soldered the analog l9r cable to my Omnibus F4 Pro RSSI pin. I also turned the telemetry funtion in the betaflight on. But when I fly my osd gives me wrong values that are between 40-60 and doesn't make sence. 

Thank you for your help in advance!

jflyper, 2017-08-13:
@Tiberius131 Did you enable "RSSI_ADC" in configuration tab?

Tiberius131, 2017-08-14:
Yes

Tiberius131, 2017-08-14:
@jflyper yes

jflyper, 2017-08-14:
@Tiberius131 I will take a look into it tonight.

jflyper, 2017-08-14:
@Tiberius131 
RSSI input pin was incorrectly defined, which was fixed with #3840.
There is still a scale problem which is under question with #3841, but you can still tweak your final output value by changing CLI variable `rssi_scale`, depending on output level of L9R at 100% RSSI.

closed 2017-08-14 by jflyper