betaflight/betaflight #3846: Any Plans to FIX 3D Mode

KenImhof, 2017-08-14:
3D mode for reversing Props was broke in 3.1.7 (works perfect in 3.1.6).  It also does not work in 3.2.0-RC3.  Motors just go to full throttle when switching directions.

Any plans of getting this fixed in the final 3.2 release?

RipperDrone, 2017-08-14:
Hopefully - it's a nice flight mode for extra fun. Wonder why nobody realized it was broken in 3.17 until now :-)

RipperDrone, 2017-08-14:
https://github.com/betaflight/betaflight/issues/3696

KenImhof, 2017-08-14:
Other people have also logged this as an issue, but appears to just be ignored.  I had to switch to iNAV earlier this year just so I could continue 3D flight without going back to an old software build.

brycedjohnson, 2017-08-14:
I had a self build version of pre-3.2 that worked.  I was experimenting with adding 3d on a switch while in 3d mode on the escs (not with dshot commands that I later did).

https://www.youtube.com/watch?v=BZ0wrvzBMwc

Not exactly the same as your issue, but I at least had it somewhat working.   It still didn't feel exactly right compared none 3d mode... so there still probably is an issue...

3d mode enabled on escs?

mikeller, 2017-08-14:
Duplicate of #3696.

closed 2017-08-14 by mikeller