betaflight/betaflight #3847: VTX Pitmode on a mode switch?

brycedjohnson, 2017-08-14:
Just was at multigpIO and though about a couple ideas that might be useful, but wanted to get a little feedback before I messed around with it.

Smart audio or tramp enable/disable pitmode on a switch?   When that option is selected then it is modes the vtx over to Race mode instead of freestyle mode (https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio).  So that the TBS will start up always in pit mode.  Flicking the switch will enable/disable pitmode.  This can be done as a OSD, but maybe useful for non OSD FCs or other times when quick ON/OFF change are needed, such as in a team race to disable a downed quad so another can go up.

Could there be an issue where the pitmode gets disabled/enabled during bootup that would cause a quick blast on that frequency?

mikeller, 2017-08-14:
There is already discussion going on about this in #3094. Closing this as a duplicate to keep the discussion in one place.

closed 2017-08-14 by mikeller

vkolotov, 2018-08-24:
I've got Akk ultimate VTX that I can switch to PIT mode via OSD, however PIT mode does not work on mode switch (though a radio switch).