betaflight/betaflight #3848: Only allow 2 props to spin in anti-turtle mode

brycedjohnson, 2017-08-14:
Just was at multigpIO and though about a couple ideas that might be useful, but wanted to get a little feedback before I messed around with it.

I believe KISS has it implemented this way and a couple people had asked about it.  With Dshot reverse motors mode enabled, only two motors would spin based on where the Pitch and Roll sticks are at.  Yaw and Throttle would not do anything.  At pitch/roll center stick perhaps no motors would spin?  I had some concerns that this could be a finger cutter if people enabled it by accident and bumped the pitch/roll sticks and two motors spun up.  

Or perhaps two motors would always spin at min throttle if at center stick?  (for example, Roll stick of 1500-2000 would spin one way and 1000-1499 spin the other, so there is no center stick where no motors would spin).

Arm may or may not need to be required?  The mode dshot reverse motors would act as the arm?

Thoughts?

robert-hoffmann, 2017-08-16:
I think detecting quad position based off gygo and then powering corresponding motors sounds the most logical to me, if we actually want to prevent spinning blades that are probably touching grass and that won't move anyways.

Even just a quick burst to one motor (if laying diagonaly on the ground) should probably flip it around.

robert-hoffmann, 2017-08-16:
Maybe the fc could do a (pretest) to see which motors it's actually capable of spinning too ?

richard-scott, 2017-08-16:
Or, it could listen to your pitch/roll input from the TX and only flip the way you ask it to go:

- No Stick input, go straight up.
- Roll right, it flips right.
- Pitch up and it flips front upwards.

Yamaford, 2017-08-16:
Another advantage is not burning motors due to the lower props touching the ground. Acc could just tell the fc what two motors are the highest. This would be the easiest two to spin up and cause a flip. No need to try and decide ourselves if the FPV camera is facing the dirt.

oxivanisher, 2017-08-22:
I also think the idea of using the giro like proposed by @robert-hoffmann is a good one. But all this "only" helps if the copter lies on the ground. What about if the copter is stuck in a tree and you try to wiggle it out with reversed motors? Then the gyro would be counter productive and the "stick idea" would be better. (Also I like the idea of the pretest!)

What about separating this into two modes:

1. Turtle - flip it if it lies on the ground
* Use gyro to detect correct motors, overwritten by sticks
* Also, it should be possible to use only one motor, if the gyro detects i.e. > 30° on pitch and roll.
* If possible pretest which motors are able to turn
2. Rescue - help wiggling it out of a tree (and hopefully not more into the tree :grin:)
* Just reverse all motors
* If possible pretest which motors are able to turn

brycedjohnson, 2017-08-24:
I'm an working on the simplest path right now, only spinning the motors commanded by pitch and roll.

JonnyPhenomenon, 2017-08-24:
Oh man, I was totally thinking about something like this last night.  I know this comment is unhelpful, but I really hope we can make this work...

Adminiuga, 2017-08-25:
But Gyro does not indicate position of the quad or which motors are touching ground/highest. Gyro indicates only angular speed which is not useful for this application.
We could use accelerator, but accelerator is known to lose its orientation after a hit.
So my vote is to use "stick position" in deciding which motors to fire up, essentially letting up the operator decide which motors to run.

closed 2017-09-07 by brycedjohnson