betaflight/betaflight #3852: Missed packet count on osd

brycedjohnson, 2017-08-15:
The other thing I was thinking about on the long drive to multigp.... Did we have a way to know if the rx misses a packet?  So the sbus packets come in every 9ms.  If the rx misses a packet

mikeller, 2017-08-15:
First thought is that this should ideally be encoded in the RSSI, to return a value for 'link quality'. Having to keep track of RSSI *and* packet loss is harder than it has to be.

bkeinert, 2017-08-15:
I would really like to have both. It doesn't have to be complicated. Just show the real absolute number of lost/bad packets (maybe reset on arm?) - computing some kind of actual RSSI value like e.g. DORA does is IMHO not necessary.

This would be helpful when analyzing "strange behavior" from just the DVR footage and you probably get an idea how many packets are actually lost.

brycedjohnson, 2017-08-16:
I'll start with at least a packet count that appears in one of the debug OSD numbers.  If it is useful we can go from there on encoding it as a RSSI.  

For encoding you'd need some steep curve.  By the time you start losing packets, you are probably just on the edge of going out of range.

closed 2018-06-23 by brycedjohnson