betaflight/betaflight #3853: 3.2 RC3 External Barometer connected MA F-405 AIO not working connected I2c

TimothyGold, 2017-08-15:
I am trying to add an external barometer (BMP180) to the Matek F-405-AIO over I2c and for some reason it is not working.  I think it may be because it is not added into the F405 config.  In other forums with other boards they seem to be able to add this to other boards.  Is there anyone that could provide more information on this and let me know if this is something that can be accommodated in BF 3.2 for this board?

avidtech, 2017-08-15:
I believe the BMP180 support isn't present until one of the later nightly builds, but that might only be for the F4 boards. I just had an issue with mine, but I haven't compiled the latest since @jflyper made changes.
Try compiling from the latest source.
Reference: #3605 #2937 #1582

jflyper, 2017-08-15:
@TimothyGold Try the latest continuous builds.

mikeller, 2017-08-15:
@jflyper: You seem to know something that I don't, or you have the wrong board in mind. There is no definition for I2C or baro for the MATEKF405 in the repo.

Anyhow, I have added I2C and baro support to MATEKF405, see the attached file. The pins to be used (for now, can possibly be improved) are TX3: SCL, RX3: SDA.

[betaflight_3.2.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1224797/betaflight_3.2.0_MATEKF405.zip)

Please let me know if this works.

jflyper, 2017-08-15:
@mikeller Dah! I didn't realize the MATEK F405 does have its own target. My bad 😢

TimothyGold, 2017-08-15:
Thanks for the updates.  I will try the build provided and provide an update.

TimothyGold, 2017-08-15:
I tried to install the build provided but I still don't see any baro.  I do see I2C error stay at 4.  Not sure if this is an indication of anything.  The build is now defaulted so I am starting from scratch.  I noticed that the I2c baro address was 0 and I searched to set the real address which I think is 119.  This did not help.  Any further recommendations?  Thanks again for your help.

jflyper, 2017-08-15:
@mikeller Can I see the target.h for the MATEKF405?

jflyper, 2017-08-15:
@TimothyGold It depends on how @mikeller configured it. Hang-on.

mikeller, 2017-08-15:
```
// *************** Baro **************************
#define USE_I2C
#define USE_I2C_DEVICE_2
#define I2C_DEVICE              (I2CDEV_2)
#define I2C2_SCL                PB10       // shared with UART3 TX
#define I2C2_SDA                PB11       // shared with UART3 RX

#define BARO
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_BMP085
#define BARO_I2C_INSTANCE       (I2CDEV_2)
```

But according to slack, I2C support is broken in current master, so hang in there while we are fixing...

jflyper, 2017-08-16:
@mikeller I2C support is fine with OMNIBUSF4 and DYSF4PRO, with an external BMP180 on I2C2 substituing UART3 over PB10 and PB11. There is obvious reason it doesn't work on MATEKF405.

mikeller, 2017-08-16:
@jflyper: Trying to get my hands on a schematic for the board, to check if there are unobvious reasons, and find a working solution.

mikeller, 2017-08-16:
Ok, my bad. I did not take into account that alternative pins are used for UART3 on this board.

Unfortunately that does not leave any good options for I2C:

```
// *************** Baro **************************
#define USE_I2C
#define USE_I2C_DEVICE_2
#define I2C_DEVICE              (I2CDEV_2)
#define I2C2_SCL                PB6        // MCU pin 58
#define I2C2_SDA                PB7        // MCU pin 59

#define BARO
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_BMP085
#define BARO_I2C_INSTANCE       (I2CDEV_2)
```

i.e. you'll have to solder onto the MCU pins 58 and 59 to get access to I2C.

[betaflight_3.2.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1226936/betaflight_3.2.0_MATEKF405.zip)

TimothyGold, 2017-08-16:
There are i2c (scl,sda) solder pads on the board. These must be mcu 58,59. Is this not correct?

mikeller, 2017-08-16:
@TimothyGold: Either I have gone blind, your you and I have different versions of the board. I have turned over my board a number of times, and examined all pads that I could find, but none seemed to be SDA / SCL. Do you have a picture?

jflyper, 2017-08-16:
I2C2 over PB10 and PB11 works if UART3 is left unconfigured (no function assigned).

TimothyGold, 2017-08-16:
[image]
This is the aio board. I2c is on the bottom.

jflyper, 2017-08-16:
Oh, then **I2C1** over PB6 and PB7 should work.

TimothyGold, 2017-08-16:
Thanks again for your help. I will give it another try.

mikeller, 2017-08-16:
My bad, the board that I've got is an F405_OSD, not F405_AIO.

Now I am totally confused, since there is only one target definition for one `MATEKF405` in the repository, and it states neither _AIO or _OSD.
(Since the difference seems to be only that the _AIO has an integrated PDB / current sensing shunt, it is entirely possible that they are both using the same target.)

@jflyper: According to the target definition, UART3 is on PC10 / PC11, so the pins for UART3 can not be used for I2C.

@TimothyGold: Try the second custom target that I attached above with the I2C pads and let me know if that works for you.

jflyper, 2017-08-16:
@mikeller I2C2, if configured over PB10 and PB11, will be activated and works **if no function is configured on UART3**. I have a working case for OMNIBUSF4 and DYSF4PRO.

@TimothyGold @mikeller You need a firmware  I2C1 configured.

TimothyGold, 2017-08-16:
I tied the second target and still no go on the PB6, PB7 resources.  I don't see the I2c error anymore.  This is 0.

mikeller, 2017-08-16:
@jflyper: I would assume on this board I2C2 will work even if UART3 is in use, since UART2 uses PC10 / PC11, and not PB10 / PB11. But to use I2C2 on PB10 / PB11, the user will have to disable OSD, and cut the trace to PB10, which is used as OSD_CS.

And yes, you are right in that the custom build above was using I2C2 instead of I2C1, here is a fixed version:

```
// *************** Baro **************************
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C_DEVICE              (I2CDEV_1)
#define I2C1_SCL                PB6        // MCU pin 58
#define I2C1_SDA                PB7        // MCU pin 59

#define BARO
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_BMP085
#define BARO_I2C_INSTANCE       (I2CDEV_1)
```

[betaflight_3.2.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1226992/betaflight_3.2.0_MATEKF405.zip)

jflyper, 2017-08-16:
@mikeller Ah, totally different mapping.

> And yes, you are right in that the custom build above was using I2C2 instead of I2C1, here is a fixed version:

This should do the job.

mikeller, 2017-08-16:
@jflyper: For the board that I've got (F405_OSD), there is an option to run baro on I2C3 (m4, m6). Since both of the boards are using the same target definition, is it possible to leverage the new, configurable I2C code to get a definition that allows for I2C1 and I2C3 to be used for baro (not in parallel, obviously)?

TimothyGold, 2017-08-16:
The last target provided does now show baro up.  I show i2c error 2 and there seems to be a problem with the leds now. No LED function at all.   UART 3 is free.  The only UARTs I am using at this point is 5 for Run Cam split control and UART2 which I think SBUS uses.

TimothyGold, 2017-08-16:
Disabling the Baro lets the leds work again.  Do you think the Baro and the Leds share the same 119 address?

mikeller, 2017-08-16:
LEDs as in LED_STRIP? If so, the bad news is you'll have to disable that feature, since I2C1 uses the same pin (PB6) as LED_STRIP.

Alternatively, you can try the below binary, and use S4 / S6 (requires remapping of S4 to S5).

```
#define USE_I2C
#define USE_I2C_DEVICE_3
#define I2C_DEVICE              (I2CDEV_3)
#define I2C3_SCL                PB41       // S4
#define I2C3_SDA                PB40       // S6
#define BARO_I2C_INSTANCE       (I2CDEV_3)

#define BARO
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_BMP085
```

[betaflight_3.2.0_MATEKF405.zip](https://github.com/betaflight/betaflight/files/1227036/betaflight_3.2.0_MATEKF405.zip)

TimothyGold, 2017-08-16:
Wow I guess maybe the baro idea was futile.  S5 is used for telemetry and s6 is used for a servo.  I think i'm asking too much from the board.  Is it possible to use UART 3 pins for I2c like @jflyper indicated?

mikeller, 2017-08-16:
No, unfortunately on this board, the pins used for UART3 (PC10 / PC11) can not be used for I2C. You could potentially move telemetry to a different UART, and then move LED_STRIP to S5, and use the SDA / SCL pads.

TimothyGold, 2017-08-16:
OK.  I will work to test this out.  If I use a straight UART TX for telemetry I think I have to tap the FRSKY non-inverted S-Port.  Is this correct?

mikeller, 2017-08-16:
Yes.

TimothyGold, 2017-08-16:
Thanks again for all your help.  One more question.  Will the I2c support make it into the final release of 3.2 for this board?

mikeller, 2017-08-16:
If we manage to work out a config and get it tested then yes.

TimothyGold, 2017-08-16:
Ok. Thank you.

TimothyGold, 2017-08-16:
I was able to move non inverted Telemetry to UART4 TX and then move LED_STRIP 1 to S5.  Everything seems to be working fine at this point.  In Betaflight the I2C error went from 2 to 1.  Not sure what this indicates.

mikeller, 2017-08-17:
@TimothyGold: Sorry, I did not quite get the meaning of your message. So you've moved LED_STRIP to S5, and then connected the baro to SDA / SCL? Now, using the firmware from https://github.com/betaflight/betaflight/issues/3853#issuecomment-322658179, the baro still is not recognised?

TimothyGold, 2017-08-17:
Sorry for the confusion. I am using the target from post where you stated you moved it from i2c2 to i2c1. Everything is working after leaving the baro on SDA SCL and moving the led strip off of the pb6 (the original led pad) and to thre s5 pin like you recommended. The i2c error 1 shows up in betaflight configurator but baro and led still working non the less.

mikeller, 2017-08-17:
That's good success. I'll submit a PR for I2C to be added then. The error is probably thrown during the initialisation.

closed 2017-08-17 by martinbudden

alfa79, 2019-01-04:
Hello, I have the same problem, the flight controller dys f4 pro v2 does not recnoce me the barometer nor the compass.
Possibly it is a configuration problem in the betaflight CLI. I see that the i2c ports are in none, also the baro and mag commands are in none.
The rest of the commands for baro and mag I do not know what values to carry.
Has anyone been able to solve these problems with said controller?
the external barometer is bmp 280 and the compass is the one that comes with the gps bn-880.
Thank you.

jflyper, 2019-01-04:
@alfa79 Your problem seems to be a configuration issue.

**Important: This is not the right place to get help with configuration issues.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

PR #3883 (2017-08-17): Added I2C / I2C baro support to MATEKF405.
Fixes #3853.