betaflight/betaflight #3855: 3.2.0 RC3 - inverted gyro corrections or something super wrong??

DEvil0000, 2017-08-15:
I tried to fly with 3.2.0 RC3 - using angle mode. I am using a eachine minicube and a motor setup for 1 LF, 2 RF, 3 RB, 4 LB (motors reassigned) reversed (front props spin away from camera). I configured everything for this. I left the PIDs and rates untouched. I tried to change filters to FIR but since this is not working well I changed it back. I also left gyro at 8kHz and PID loop at 2kHz (18%CPU). I double/tripple checked everything is correct. It is also calibrated correct.
When starting the motors the copter is stable for a short while then it starts pitching nose up and rotating with yaw even bevore I can hover. With a bit more thrust it flips backwards. I tried to compensate but then it just flips somewhere else.
For me it seams like the gyro corrections are applied to the wrong motors (basically inverted).

Arakon, 2017-08-15:
Considering absolutely noone else has this issue, it's a setup problem and not a bug. You probably have the orientation wrong or the motor order is wrong.

DEvil0000, 2017-08-15:
I checkt all of those 3 times and now again. They are correct.Maybe I will do a reset and try it again later.

jflyper, 2017-08-15:
If props are spinning reverse, mmix must be modified accordingly to get the correct yaw motion.

RipperDrone, 2017-08-15:
maybe command for reversing motor direction was not ported over to BF 3.2 correctly since name of the parameter has changed: Before: set yaw_motors_direction = -1. New in BF 3.2: set yaw_motors_reversed = ON.

wind0r, 2017-08-15:
There wasnt any "Port". only the name changed!
For me this totaly sounds like wrong motor mixer / motor order!

RipperDrone, 2017-08-15:
By "port" I mean: Trying to import CLI saved settings from older version BF into BF3.2. Doing so, the name change might result in motor control still NOT being reversed as intended :-)

wind0r, 2017-08-15:
Ahh! Import :)
So we both agree that it looks like a Usererror :) I first thought you are searching the issue in Betaflight Firmware.

wind0r, 2017-08-15:
Posting a diff would help here :) but it also sounds more like a support request and not a issue :)

RipperDrone, 2017-08-15:
mixed up yaw_motors_reversed and yaw_control_reversed switches? Just going through some guessing here...

DEvil0000, 2017-08-15:
I agree it looks sounds like a user/configuration issue but i double/tripple checked everything and I did it the same way some days before with 3.1.7 where it worked perfectly.

DEvil0000, 2017-08-15:
I did not reset the config when installing the new version. But all settings were lost so I tried to import the backup but the settings were still lost. So I was setting it up from scratch without a reset. And again all settings have been correct and checkt very very often. It rotated quite strange mostly by yaw (like its wrongly configured but no matter how I configured it it was not working correct). Maybe some old config value had some influence?? However now I was resetting the config and doing it again from scratch and now it works as expected so I close it. But I now think its some bug in config migration/import in combination with inverted yaw..

closed 2017-08-15 by DEvil0000