betaflight/betaflight #3857: Betaflight OSD is getting CRAZY !

Bugg69s, 2017-08-15:
I bought my self Omnibus f3 ( from aliexpress. It was just 15$) when i first have connect camera and vtx to it i can see normally. BUT ! after 30s its get crazy. Text is changing to simbols scrren is full of random zeros. And then you see just zeros and sometimes simbols. Plz help me. I just wanth fly my quad. Its my first build im building it like 5 months. I always burn something and now i dont wanth to change FC becouse i will burn it. Its good that i cant smoke the frame. 😶 PLZ HELP ME !

Bugg69s, 2017-08-15:
Photo: https://drive.google.com/file/d/0B1_uE11Nn9LDN0Q3TUFGNG5VbWM/view?usp=drivesdk

DieHertz, 2017-08-15:
Your ticket looks like a support request.
I'm sorry to inform you that GitHub is a place for development and reporting bugs, please redirect your request to the RCGroups.com forums thread or to your hardware supplier.
Also bear in mind that getting cloned hardware is generally not advisable as it usually contains flaws and differences from the reference.

Bugg69s, 2017-08-16:
@jflyper i know that have clone is not the best. But i dont wonth to pay 30 dollars for FC + 20$ shipping. Its not worth for me. I will burn it. I dont have that soldering skill.

jflyper, 2017-08-17:
Suppose you successfully managed to finish your build without burning something.

Then you will end up with a craft that only flies for 30sec and your are back to the beginning. What do you think?