betaflight/betaflight #3861: MS5611 baro shows random ~1-kilometer jumps

andrejpodzimek, 2017-08-16:
After a bit of tweaking, I ran into quite a severe baro issue on a BlueJayF4 board ([the FlyMaster clone](http://www.helipal.com/flymaster-fmf405-f4-flight-controller-bluejayf4.html)).

After I [got my baro and compass working](https://www.rcgroups.com/forums/showthread.php?2593106-BlueJayF4-STM32F4-based-FC/page70#post38028235) in Betaflight, I had to solve [a faulty GPS issue](https://github.com/betaflight/betaflight/issues/3777). I got a [different GPS device](https://www.banggood.com/NEO-M8N-Flight-Controller-GPS-Module-GYGPSV5-NEO-for-Pixhawk-APM-p-974362.html) with an HMC5983 built-in magnetometer. So instead of the separate I2C magnetometer of the same type that I had been using, I simply switched to (i.e., connected the I2C wires to) the magnetometer built into the GPS sensor. Compass works just fine and so does GPS (on UART).

The barometer is soldered onto the same I2C wires as the compass. Both sensors are detected, but the barometer is making random jumps to >800 meters of (positive) altitude. The jumps are accompanied by I2C errors. In comparison to the previous working configuration (no jumps, 0 I2C errors), the I2C connection scheme is the same, the compass is of the same type and the baro is even the very same device. I used to see zero I2C errors all the time, but now there are ~tens of them within a few minutes of uptime and huge baro jumps every now and then. Jumps get way more frequent when the quad is armed or flying.
[image]

The only thing that's (electrically) different is the GPS power supply. The previous one was UART-only (no compass) and explicitly said it wanted 5V. The new one says it wants 3.3V, just like I2C devices, although the GPS part is on UART. So I soldered it to the 3.3V power pad for Spektrum receivers.

What could be wrong here? Is it related to cycle time? Could it be an issue with the length of wires? They got a few cm longer, but this is a 3" hexacopter, so there are no really long onboard distances. I'd be more than happy to try some debugging and troubleshooting tips.

jflyper, 2017-08-16:
I think it's related with I2C bus transaction errors you are experiencing with #3777. The code does not care the I2C device transaction status, whether it was successful or not, when reading measurements.

I will be looking into the code, but it will not give you the correct reading even the ignorance is fixed. (I will probably add a code at a driver level that reports last successful measurement if I2C fails.)

You have to find out why you are getting the errors. I suspect cable, pull-ups being too strong or too weak ... scoping it with DSO is the best way to find out why.

digitalentity, 2017-08-16:
I suspect it might be related to I2C bus speed. Recently I've stumbled upon several (cloned) samples of MS5611 and HMC5883L that don't tolerate 400kHz clock speed. Reducing speed to 200kHz seems to eliminate errors.

andrejpodzimek, 2017-08-16:
Is there an option for setting the I2C frequency? It doesn't have to be in CLI; I'm running master anyway, so I can just rebuild it with a suitable tweak or pull request.

The baro is probably OK, because it worked 100% fine with the previous instance of HMC5983. The current new HMC5983 instance (the one inside the new GPS) is the only (big physical) difference in the setup. But admittedly, my I2C wires are now longer than before.

Also I was wondering if the 3V onboard regulator could be insufficient for powering all 3 devices (UART GPS, I2C baro, I2C magnetometer), but I didn't spot any voltage drops that would correlate with the baro jumps.

andrejpodzimek, 2017-08-16:
One more observation: When the drone just sits on USB and/or battery, it has ~2 baro jumps per minute on average. But once I take off, the baro jumps occur almost all the time and some of them "reach" up to 3 kilometers. (I can watch the altitude on the transmitter via SmartPort telemetry, which is how I discovered this...)

jflyper, 2017-08-16:
@andrejpodzimek It is all related to the inconsistent I2C connection.

andrejpodzimek, 2017-08-16:
I couldn't reproduce it by shaking it. But arming/disarming definitely increases the frequency of jumps and so does connecting the battery. Movement doesn't seem to have an efffect.

jflyper, 2017-08-16:
You really have to scope it, if you want a usable mag.

jflyper, 2017-08-16:
Oh, BTW, the I2C driver (or GPIO in general) had a fatal bug in ~RC3. You may want to wait for RC4 before you do some further fiddling with mag/baro on I2C.

andrejpodzimek, 2017-08-16:
I've just built `BLUEJAYF4` with `I2C{1,2,3}_OVERCLOCK` set to `false`, out of curiosity. That made the situation much worse, surprisingly. Mag didn't show anything at all (stuck at min/max values) and baro made >2km jumps all the time. And there were hundreds of I2C errors per minute rather than tens. OK. Something else is wrong. I'll try to shorten the wires as much as possible. It's a shame that I don't have an oscilloscope and know ~nothing about electronics.

andrejpodzimek, 2017-08-17:
Back to an unmodified master build.
I re-wired the I2C to make the wires as short as possible, but things still look broken.
[image]
Baro jums are accompanied by 1 I2C error each. Compass noise doesn't correlate with I2C errors. It looks like the last resort would be to restore the old compass, completely separate from the GPS. :cry:

andrejpodzimek, 2017-08-17:
I’ve ordered an oscilloscope. Will report back once I get it.

andrejpodzimek, 2017-08-20:
Got [this oscilloscope](https://www.picotech.com/oscilloscope/2000/picoscope-2000-overview), the 100 MHz version.

The first surprise was the need for a x10 probe attenuation. With x1 it just breaks the I2C communication somehow. (Whenever I touch SDA with the probe, it causes tens of I2C errors per second. When I touch SCL, it still causes I2C errors, although fewer, units per second.) I haven't seen an I2C error yet while using the x10 attenuation.

Thus far I've only experimented with the compass and without the barometer, just to get a basic idea how an oscilloscope works and what I should see on the bus.

In general, what should I be looking for? I plan to connect the barometer back and reproduce the clash between the two devices under oscilloscope measurement. But first of all I'd just like to check whether what I've just measured looks OK for an I2C bus.

There are short pulses ~10 times per second on both SDA and SCL and there's a lot of structure when I zoom into a pulse.

The SDA signal, bird's eye view:
[image]
The SDA signal, zooming into one of the ~100µs pulses:
[image]

The SCL signal, bird's eye view:
[image]
The SCL signal; zooming into one of the ~100µs pulses:
[image]

digitalentity, 2017-08-20:
If your scope has indeed 100MHz bandwidth - you have a very shitty signal on your i2c bus - I'm surprized it works at all.

andrejpodzimek, 2017-08-20:
Yes, the scope has a 100 MHz bandwidth. I'm sad to hear the signal is shitty :cry:, but what can I do about it?

I already tried to disable the I2C overclocking, but then none of the two devices worked.

I also tried to add extra resistors (10 kΩ) between 3 V and SCL as well as between 3 V and SDA (to decrease the total pull-up resistance), just guessing from the fact that SCL as well as some short SDA pulses don't make it above 3 V at all. But that didn't work as expected. Both devices got recognized, but the number of I2C errors was huge and the data from both devices was just bogus.

Also tried to disable/re-enable internal pullups, which didn't seem to have any effect on the number of errors etc. (Either it doesn't apply to this MCU or perhaps the compass already has pullups built in. I expected I2C to just fail without the internal pullups, but that wasn't the case.)

Anyway, suggestions around further hacks and tweaks as well as magic ideal pullup resistances are more than welcome.

jflyper, 2017-08-20:
Internal pull-ups does nothing for anything. It may require the external pull-ups to to as small as 2K combined.

digitalentity, 2017-08-20:
I would go to 1K pull-ups right away.

AirBreak69, 2017-08-22:
I like to push you into another direction - away from electronics, although i agree that the waveforms and the 1x probe effect show weak signals.
Probably the folowing is a new/different problem.
My setup: REVO (8K/8K/DSHOT/IBUS), naked board, nothing connected, 10K pullups. No I2C errors in general.
My observation with build 1592: 
a. disabling LED_STRIP: sporadic glitches (1 per 30s) on mag signals and sporadic jumps on baro data
b. enabling LED_STRIP: sporadic glitches (1 per 30s) on mag signals but random baro data!
With enabled LED_STRIP the max time of the baro task rises from 116us to 120us.
Going back to build 1588:
a. disabling LED_STRIP: clean mag signals and sporadic jumps on baro data
b. enabling LED_STRIP: sporadic glitches (1 per 30s) on mag signals and sporadic jumps on baro data
Max time of the baro task stable at 120us.

Can you think of corrupted data transfers due to some race condition in the tasks scheduling or DMA operation?
Reviewing the past, the stability of the mag/baro data on my REVO "firmware test FC" was depending/varying with the numerous changes in low level/drivers code.

mikeller, 2017-08-22:
@AirBreak69: That is definitely a possiblity. Unfortunately, the limitations on how much concurrent DMA the STM32 MCUs can handle seems to be a hardware limitation, and we seem to be pushing these limits with how Dshot is done: https://github.com/betaflight/betaflight/issues/2631#issuecomment-290664905

mikeller, 2017-08-23:
@AirBreak69: Good findings. The lack of synchronisation on bus access is definitely something that will have to be investigated.

andrejpodzimek, 2017-08-23:
I [noticed the overclocking](https://github.com/betaflight/betaflight/issues/3861#issuecomment-322885625), but setting the `#define`s to `false` made things way worse. Instead of a few glitches from time to time I was getting completely bogus baro data (like +/- 30 km of altitude) and the compass had twitches at least every second.

I had expected the bus to become more stable without overclocking, but that wasn't the case. If this is an interference pattern of some sort, then the bus slowdown could have (pseudo)randomly made things worse, rendering any signal stability gains useless.

I wouldn't mind running the I2C bus at a slower speed, because my board has a gyro on SPI, so IIUC, the I2C bus is not a loop time bottleneck in my case. The delays induced by reading the compass and baro would be longer (and the reads seem to be done [synchronously](https://github.com/iNavFlight/inav/issues/1931#issuecomment-322151866) in the current implementation), but they only occur 100-times or 10-times per second.

AirBreak69, 2017-08-23:
There seem to be several problems in the code:

1. Both common_fc_pre.h and common_osd_slave.h (re)define the I2Cx_OVERCLOCK-defines. I changed both but no effect.
2. The I2Cx_OVERCLOCK-defines aren't used as only drivers\bus_i2c_stm32f30x.c takes care of them. But only STM32F3.mk uses it which should result in 1000kHz(!) or 400kHz I2C clockspeed (if comments are correct). STM32F1.mk and STM32F4.mk both use bus_i2c_stm32f10x.c which results in 800kHz I2C clockspeed indepently of the I2Cx_OVERCLOCK-defines. 
I tried a dirty
    if (pDev->overClock) {
//        i2cInit.I2C_ClockSpeed = 800000; // 800khz Maximum speed tested on various boards without issues
        i2cInit.I2C_ClockSpeed = 400000; // 400khz Operation according specs
    } else {
        i2cInit.I2C_ClockSpeed = 400000; // 400khz Operation according specs
    }
Now the I2C clockspeed is 400kHz but the mag data are static(!) and baro data show the same jumps.
I'm confused and have to state not beeing deep enough inside the code to change something without side effects.

andrejpodzimek, 2017-08-23:
More bad news. :disappointed:  I connected 2.2 kΩ pull-ups to SDA and SCL. The total pull-up resistance appears to be 1.7 kΩ in both cases. **I also soldered the barometer back on.**

I'm still getting pretty much the same baro jumps and compass twitches. There are 0 I2C errors (unless I play with the probes too much), yet something is obviously wrong.

The signal shows an interference pattern where two periodic transfers (probably from the baro and the compass) approach each other and then diverge again. Some of the transfers were quite close to each other (closer than the duration of a transfer), but I never spotted an obvious clash. They get quite close and then diverge again.

I believe my signal is better with the pull-up resistors (SCL is blue this time):
[image]

When zooming in at random, I spotted surprising clock patterns in the second half of some of the transfer bursts (SCL is blue):
[image]

Zooming in:
[image]
Another similar phenomenon:
[image]

Are these clock irregularities expected? Could it be a timing issue in Betaflight? I've just let the drone sit for 30 minutes and there are still 0 I2C errors, yet the baro keeps jumping up to 3 kilometers and the compass has min/max value twiches once in a couple of seconds.

This might be related to some of the issues @AirBreak69 has mentioned.

mikeller, 2017-08-23:
@andrejpodzimek: This could be a timer issue. Are you testing with a fully configured flight controller? If so, could you test if this improves when you disable as many other timer / DMA users as possible (Dshot / LED_STRIP / UARTs / ...)?

andrejpodzimek, 2017-08-23:
It looks like two transfers don't get closer to each other than this. Then they diverge again.
[image]

andrejpodzimek, 2017-08-23:
@mikeller Yes, this is a hexacopter with D-Shot, with SmartPort on SoftSerial, GPS on UART 3, S.Bus on UART 6 and (mw)OSD on UART 1. Quite fully configured, in all meanings of the word. So I'll switch to multishot for a while and retry.

AirBreak69, 2017-08-23:
@mikeller I disabled DSHOT (changed to ONESHOT125) but left LED_STRIP enabled.
Although the I²C is still overclocked at 800kHz, mag and baro data are stable and clean now!
Now it's time for geniusses...
@jflyper currently tries an approach to disable unused DMAs.

andrejpodzimek, 2017-08-24:
@mikeller So with multishot I'm not getting any compass twitches and the baro has only jumped once in ~20 minutes (and I can't say with certainty that it wasn't just my probes touching exposed wires).

Things look similar on the oscilloscope, but not exactly the same. The clock irregularities I zoomed on above are still there, somewhat more often than before. Now I can spot ~3 clock irregularities per transfer instead of 1-2. Also, whenever SCL stops for a while, SDA stays up. Yet another difference from the previous state.

Surprisingly, I can see two different "zero" levels on SDA now. Not sure why that's the case, but all transfers I've zoomed into have that. As if one of the devices had too much internal resistance to short the bus all the way to 0, now that I have the new pullup resistors. This shows the two different zeros quite nicely (SDA is blue):
[image]

And some SCL gaps (red) with SDA staying up (blue) are also visible.

andrejpodzimek, 2017-08-24:
I've just had >20 minutes without a single glitch on multishot (which has no DMA stuff in `resource list`).

Then I switched back to `DSHOT600` and both I2C devices got twitchy. The baro is on Mont Blanc already. Anyway, I can confirm that switching dshot off improves things dramatically.

jflyper, 2017-08-24:
I can't reproduce the problem with OMNIBUSF4SD with UART3 (PB10, PB11) switched to external HMC5883 + BMP280 pair, Dshot on and off...

andrejpodzimek, 2017-08-24:
@jflyper Could the problem be specific to I2C on `PB8` and `PB9`? (Those are the BlueJayF4 I2C pins you mentioned [here](https://www.rcgroups.com/forums/showthread.php?2593106-BlueJayF4-STM32F4-based-FC/page69#post38021320).) Not sure if they're accessible on the omnibus though.

digitalentity, 2017-08-24:
The problem is probably noise-related. DSHOT is higher frequency than multishot and produces more RF noise which may be messing with I2C if it's poorly laied out on the PCB and not terminated properly.

jflyper, 2017-08-24:
I2C1 on PB8 and PB9 ... they are on ... S2 and PPM. Yes, I can do it. Hang on.

jflyper, 2017-08-24:
Doesn't reproduce with the mocked BJF4 (with fake acc/gyro) with HMC5883 and BMP280 on I2C1.

So, it happens with only the HMC5883 connected, too, correct?

andrejpodzimek, 2017-08-24:
Yes, it does. I'm getting compass twitches even without the baro when using dshot. (The compass is in fact an HMC5**9**83, but that's probably irrelevant; it should be backward compatible.)

jflyper, 2017-08-24:
Are you running GPS at the same time?

andrejpodzimek, 2017-08-24:
Yes, I have the GPS enabled. It's on UART3. It's on the same PCB as the compass, but only the compass is on I2C.

jflyper, 2017-08-24:
Let me try that with my puck.

andrejpodzimek, 2017-08-24:
Here's my `diff all`, just in case. If it contains anything relevant (other than dshot), I'll be more than happy to experiment. If some F4s have a timer issue and some don't, then there's probably no way to debug this properly. Sadly I don't have another F4 for comparison.

    # diff all
    
    # version
    # Betaflight / BLUEJAYF4 (BJF4) 3.2.0 Aug 17 2017 / 03:01:44 (677b34f6b)
    
    # reset configuration to default settings
    defaults
    
    # name
    name Raiju
    
    # resources
    resource SERIAL_TX 6 NONE
    resource SERIAL_TX 11 B03
    resource SERIAL_RX 11 NONE
    
    # mixer
    mixer HEX6X
    
    # servo
    
    # servo mix
    
    
    # feature
    feature SOFTSERIAL
    feature GPS
    feature TELEMETRY
    feature ANTI_GRAVITY
    
    # beeper
    
    # map
    
    # serial
    serial 0 1 115200 57600 0 115200
    serial 2 2 115200 57600 0 115200
    serial 30 32 115200 57600 0 115200
    
    # led
    
    # color
    
    # mode_color
    
    # aux
    aux 0 0 0 1700 2100
    aux 1 1 1 900 1625
    aux 2 2 1 1625 1875
    aux 3 3 1 900 1375
    aux 4 5 6 1700 2100
    aux 5 11 1 900 1125
    aux 6 13 3 1300 2100
    aux 7 28 4 1700 2100
    
    # adjrange
    adjrange 0 0 2 900 2100 12 2
    
    # rxrange
    
    # vtx
    
    # rxfail
    rxfail 5 s 1000
    rxfail 6 s 1000
    rxfail 7 s 2000
    
    # master
    set gyro_sync_denom = 2
    set gyro_use_32khz = ON
    set align_mag = CW180FLIP
    set mag_hardware = AUTO
    set mag_declination = 218
    set magzero_x = -203
    set magzero_y = -3
    set magzero_z = -360
    set baro_i2c_address = 119
    set min_check = 1000
    set max_check = 2000
    set rssi_scale = 30
    set fpv_mix_degrees = 26
    set blackbox_device = NONE
    set min_throttle = 1030
    set dshot_idle_value = 200
    set motor_pwm_protocol = DSHOT600
    set failsafe_delay = 10
    set bat_capacity = 1000
    set current_meter = ADC
    set ibata_scale = 366
    set gps_provider = UBLOX
    set gps_sbas_mode = EGNOS
    set gps_auto_baud = ON
    set pid_process_denom = 1
    set frsky_gps_format = 1
    set frsky_unit = METRIC
    set sdcard_dma = ON
    set camera_control_key_delay = 150
    
    # profile
    profile 0
    
    set dterm_lowpass_type = PT1
    set vbat_pid_gain = ON
    set anti_gravity_gain = 2000
    set yaw_lowpass = 90
    set i_alt = 15
    set d_alt = 25
    set d_level = 80
    
    # profile
    profile 1
    
    set dterm_lowpass_type = PT1
    set vbat_pid_gain = ON
    set anti_gravity_gain = 2000
    set yaw_lowpass = 90
    set i_alt = 15
    set d_alt = 25
    set d_level = 80
    
    # profile
    profile 2
    
    set dterm_lowpass_type = PT1
    set vbat_pid_gain = ON
    set anti_gravity_gain = 2000
    set yaw_lowpass = 90
    set i_alt = 15
    set d_alt = 25
    set d_level = 80
    
    # restore original profile selection
    profile 0
    
    # rateprofile
    rateprofile 0
    
    
    # rateprofile
    rateprofile 1
    
    set roll_srate = 80
    set pitch_srate = 80
    set yaw_srate = 80
    
    # rateprofile
    rateprofile 2
    
    set roll_srate = 85
    set pitch_srate = 85
    set yaw_srate = 80
    
    # restore original rateprofile selection
    rateprofile 0
    
    # save configuration
    save

jflyper, 2017-08-24:
I have loaded your `diff` output; still can't reproduce.

I suspect (another) hardware problem with your puck...

jflyper, 2017-08-24:
> I've just had >20 minutes without a single glitch on multishot (which has no DMA stuff in resource list).
>
>Then I switched back to DSHOT600 and both I2C devices got twitchy. The baro is on Mont Blanc already. Anyway, I can confirm that switching dshot off improves things dramatically.

This is the weirdest part...

mikeller, 2017-08-24:
Could it be that limitations on the maximum number of concurrent DMA operations causes corruption of the I2C data?

digitalentity, 2017-08-24:
There is a documented errata on DMA controller for F4, but it only affects DMA operations, while I2C doesn't use DMA.

mikeller, 2017-08-24:
@digitalentity: Ok, I wasn't aware that I2C does not use the DMA controller.

AirBreak69, 2017-08-24:
@andrejpodzimek You don't look like an electronic newbee. Hope not harming you with this thoughts:
Different levels normally mean "driven by other device" or "more than one device try to drive competing logical levels".
It would have been very usefull if you purchased an extended version of the pico scope. The I²C interpreter could tell us if the low level shift occures in ACK/NACK slot where the slave device could send "i did not understand". But NACKs should lead to I2C errors.
One had to study https://en.wikipedia.org/wiki/I%C2%B2C - if it weren't too late now ;-)

AirBreak69, 2017-08-24:
@andrejpodzimek Don't now if it's still relevant:
The board is configured as a simple quad. No exotic mapping.

name REVO_ZR220
# resources
# mixer
# servo
# servo mix
# feature
feature GPS
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
# beeper
# map
# serial
serial 0 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
# led
led 0 1,0::A:6
led 1 2,0::L:6
led 2 3,0::CW:6
led 3 4,0::G:0

ledvinap, 2017-08-24:
There was some timing limitation on F1, which needed fast IRQ handling. I did fix it in my local branch, and it was probably merged into CF v1.x, not sure about BF .. 
https://github.com/PX4/NuttX/issues/54

andrejpodzimek, 2017-08-24:
@AirBreak69 In fact I know ~nothing about electronics; my specialization (as in work + education) is computer science and software (not for MCUs and UAVs though). This is the very first time I had a closer look at something "low-level".

I don't think I need a more advanced Picoscope to do logic analysis. That [seems to be supported](https://www.picotech.com/library/oscilloscopes/serial-bus-decoding-protocol-analysis) with all Picoscope models; I just haven't had time to get familiar with it. Sadly, the Picoscope software is a bad disappointment. It’s something closed-source ([I installed it from this package](https://aur.archlinux.org/packages/picoscope/)) and full of UI glitches. :disappointed:

The best way to go (for the time being) may be to use the Picoscope software only to save traces in full resolution and feed them into a separate analyzer software, [such as Sigrok](https://sigrok.org/). At this point Sigrok can't read from Picoscope directly, but this appears to be [on the roadmap](https://sigrok.org/wiki/Supported_hardware#Oscilloscopes).

andrejpodzimek, 2017-08-24:
The fact that switching dshot off makes it work is somewhat puzzling. (Needless to say there are many reasons to prefer dshot over the analog protocols.) Speaking of timing, the `DSHOT600` beeps are out-of-sync in about 1/3 of cases when I plug in the battery. As if some ESCs (2 at least) booted later than the rest of them. The difference is less than a second, but clearly audible. In ~2/3 of cases they are in sync, as far as my ears can tell. I haven’t paid much attention to this, but now I thought I should mention it in this context.

andrejpodzimek, 2017-09-01:
As for the dshot —> multishot transition, I spoke too soon. Baro jumps are still there with multishot. 😢 They just became infrequent. The drone can sit on the table for an entire battery without a single glitch, but the baro jumps do happen in flight. I set an alarm on my TX to beep when the telemetry altitude is above 200 m. And it beeps very often, many times per flight. (With it comes the risk of an FC crash and fall.)

So, multishot makes the baro jumps less likely to occur, but the underlying problem still persists.

jflyper, 2017-09-01:
@andrejpodzimek Does your current firmware has `I2Cx_OVERCLOCK` defined as `true`? If so, redefining it to `false` may change the way sensors behave on I2C bus.

andrejpodzimek, 2017-09-01:
@jflyper These lateste flights were on a master build from unmodified sources, i.e., whatever `common_fc_pre.h` says is used (I think all 3 buses available on `STM32F4` are overclocked). I did try to disable I2C overclocking [a few days ago](https://github.com/betaflight/betaflight/issues/3861#issuecomment-323579298), but that just failed (neither mag nor baro got detected and the status bar showed thousands of I2C errors), so I reverted that hack.

jflyper, 2017-09-01:
Try it again. If it makes things worse, your I2C bus is really screwed up.

andrejpodzimek, 2017-09-01:
Done. With overclocking set to `false`, I'm getting ~10 I2C errors per second and the FC crashes within a few minutes (i.e., becomes unresponsive) even while sitting on the table. With overclocking enabled I'm getting only a handful of I2C errors (~1 per minute, maybe). But the baro glitches are no longer directly correlated with I2C errors. Both phenomena seem to occur independently of each other now.

andrejpodzimek, 2017-09-03:
This must have been a combination of two orthogonal problems. Interference with dshot was one of them; disabling dshot helps. But it doesn't solve the problem completely; something else is needed. I have made a hardware modification. My I2C bus looks different now (higher tension, two levels of "zero" SDA again, this time even further apart). And I've just had an entire hour on the table with zero I2C errors or twitches and a few flights without issues.

[image]

This time I don't want to speak too soon. I'll take a deep breath, more flights (~10 batteries at least) and then report back the details of the latest hardware tweak.

It would be great to have it all working with dshot (and perhaps #4020 can help by simply side-stepping whatever timing bugs there might be right now), but multishot is not a bad solution for the time being. (BTW, the CPU load at 16 kHz loop frequency is 47% with dshot and only 25% with multishot.) Now I can finally get back to #3823 and look if I can find altitude PID values that give me a stable hover.

themelle, 2017-09-06:
I have been building by quad with GPS, mag and baro always on the must-have list, and was glad to get GPS (Beitian BN-880) working with BF3.1.7.
Now with BF3.2 even mag (HMC5883L on GPS module) and baro (MS5611-01BA03) are being recognized! 
GPS is working pretty good, but I do get the same 800m jumps in baro height as Andrej reported, although at a much higher rate (about every 3-5sec). 
Baro and mag are connected in parallel to i2C at UART3 on an OMNIBUSF4 board. 

I am using DSHOT600 and running 8kHz PID and gyro loop.

What is this hardware modification you mentioned in your previous post, Andrej? Maybe I can give it a try ... ? :-)

andrejpodzimek, 2017-09-06:
@themelle It looks like certain devices get on well on the bus and some don't. Maybe I'll start with a summary of what I've tried. It all started with [these](https://www.banggood.com/3Pcs-GY-282-HMC5983-High-precision-High-sensitivity-Temperature-Compensation-p-1144414.html) [two](https://www.aliexpress.com/item/MS5611-01BA03-SENSOR-new-and-original-free-shipping-500BP/32460242247.html) parts:

[image]

They were both on 3.3 V on the same bus and worked just fine. I didn't notice any glitches or similar issues and I even used them together with dshot.

Then I got [a different GPS](https://www.banggood.com/NEO-M8N-Flight-Controller-GPS-Module-GYGPSV5-NEO-for-Pixhawk-APM-p-974362.html) with a built-in compass (because the previous one [didn't work](https://github.com/betaflight/betaflight/issues/3777)). I ditched the external compass and started using the one in the GPS, which is when things went sideways.

  * Both on 3.3 V worked, but had I2C glitches and baro jumps.
  * Baro on 3.3 V and compass + GPS on 5V (which is OK, behind LDO circuit) had jumps and glitches as well, even more frequently.
  * Both scenarios combined with 2.2 kΩ pull-up resistors were pretty much the same; the signal looked better on the oscilloscope, but things didn't improve much otherwise.
  * That concludes the list of options, because the baro was not behind a LDO regulator.

Whatever I tried, I still had compass and baro twitches. It was ~1 jump per minute without dshot and 10+ glitches per minute with dshot enabled. I have no idea what was causing this correlation, but that's just how it was. There were also lots of I2C errors.

OK, so much for the history. **This is the modification that I've made (left: [before](https://www.aliexpress.com/item/MS5611-01BA03-SENSOR-new-and-original-free-shipping-500BP/32460242247.html), right: [after](https://www.banggood.com/3Pcs-MS5611-GY-63-Atmospheric-Pressure-Sensor-Module-IICSPI-Communication-p-1122870.html)):**

[image]

This is how it's actually connected to the GPS. The most important change is that the baro is now behind a LDO regulator and can be powered from 5 V, just like the compass + GPS.

[image]

With this^^^ configuration I have never seen a glitch on multishot. However, with dshot I'm still getting baro jumps (units of jumps per minute), although this time, surprisingly, they are **not** correlated with I2C errors, i.e., the number of I2C errors stays at **zero** all the time. Staying with multishot for the time being; otherwise things work fine. [Unless I want to acually use the baro](https://github.com/betaflight/betaflight/issues/3823), but that's a different issue.

themelle, 2017-10-04:
@andrejpodzimek Thanks for your thorough explanation - altough it does not really solve the problem, it may give someone with a deeper understanding a clue to find a solution.

On a side note I've just stumbled upon [PR#1934](https://github.com/iNavFlight/inav/pull/1934) in iNav. If this could be implemented in BF maybe we could have more tools at our disposal to try different things. I have high hopes that 200kHz I2C bus speed may help our setups alot ...

jflyper, 2017-10-04:
A couple of comments.

- "I2C Errors" as reported counts number of "_I2C **BUS** protocol errors_".
There may still be bit errors, which may cause huge jumps depending on where the error bit is.

- Slower I2C will provide more reliable communication. I will look into this as I overhaul the I2C for DMA toward 3.3 or 3.4.

themelle, 2017-10-05:
@jflyper Thanks for your thoughts. 
And many thanks in advance for the promised I2C overhaul - I'm looking forward to it, can't wait 👍 
If we can assist you testing your implementation please drop a line!

andrejpodzimek, 2018-01-08:
I’ve just tested DShot on the `ANYFCF7` clone of the `BLUEJAYF4` hexacopter this bug had been initially about. The new DShot timers (1 timer per 2 motors) on F7 are a.w.e.s.o.m.e. I have DShot for the first time without any I2C baro jumps, FC crashes or other mishaps. There are 3 of those new DMA resources, not 6. This may also resolve #3962 eventually, but I haven’t tested that yet.

[7 more replies]

closed 2018-01-10 by andrejpodzimek