betaflight/betaflight #3863: Prevent crash in case of motor fail

matoy, 2017-08-16:
Hello,

First, thanks to everybody for all the hard and great work!

New crash recovery feature on BF3.2 seems exciting.
After having a few bad crashes due to inflight motor failing (i guess I should not buy cheap chinese goods...), I wonder if that would be possible to add a "prevent crash" feature?

There are some threads on that subject on the internet, for example: https://techcrunch.com/2014/03/05/quadcopters-never-have-to-crash-again-thanks-to-this-software-based-fail-safe-system/
In case of motor failing, the quads "yaws" very fast; it's impossible to pilot BUT it allows to land and thus avoid a major crash.

That could be a new mode activated by a RC switch for example.
Thanks to let me know if you find this interesting.

Regards,

martinbudden, 2017-08-16:
I've thought about implementing something like that at some point. However it is of limited use in betaflight - it stabilises roll, pitch and thrust by allowing the quadcopter to spin. That's OK for line of sight flying, but doesn't help if you are flying FPV.

DieHertz, 2017-08-16:
It somewhat helps, just imagine you're a helicopter pilot :-)

matoy, 2017-08-16:
> That's OK for line of sight flying, but doesn't help if you are flying FPV.
if you're flying FPV and activate the mode quickly enough, it can help to land (by slowly reducing throttle) and avoid big crash, doesn't it?

libwtf, 2017-08-16:
i see this as quite dangerous in a freestyle/race environment....better to just go down than a spiraling mess you can't navigate..
i can see that for bigger rigs with cameraplatform and stuff - although there is much more mass moving which brings it's own problems...

andrejpodzimek, 2017-08-16:
Do hexacopters and octocopters actually have motor failure resilience in Betaflight? In their case it's perfectly feasible to land the aircraft after a motor faiure, beause there are enough other motors to keep lift and torque in balance. (In efffect, a hexacopter turns into a quadcopter and an octocopter can become a hexacopter (or a quadcopter, if it has enough thrust).)

matoy, 2017-08-16:
@libwtf : depending on the situation, people might activate or not this feature and/or launch failsafe.
many people pilot drone in empty environnement where it's okay and that would avoid many broken parts and repair.
@andrejpodzimek : that's my next project: build and hexacopter, frame is on its road ;-)

andrejpodzimek, 2020-02-12:
Has there been any recent research into this, especially considering the new bidirectional DShot? If I'm not mistaken, one of the challenging parts of failsafe control algorithms would be a continuous correlation analysis to determine which of the motors are failing. With bidirectional DShot, however, many types of anomalies would be way easier to detect. Instead of complex movements of the frame, the only thing to watch would be the "thrust to rpm" ratio.

It's just that &hellip; my octocopter lost a propeller and flipped over and crashed in the middle of my living room. There seemed to be no way to recover from the situation using control inputs, as far as I can tell from the split second between the propeller disintegration and the crash. If *something* could stop the opposite motor, that would have saved the situation.

UltiFix, 2020-05-07:
I know in the past this feature has not been super applicable to many of the smaller FPV drones many of us fly, but I think now GPS capability will continue trickling down to many copters. If you combined GPS rescue with the "motor failure flight mode" this could be super useful.