betaflight/betaflight #3869: Anti Turtle Mode won't work on Matek F405

AndyCWilson, 2017-08-16:
So I enabled anti turtle mode and on the bench it appeared to work fine with the quad upright, spinning the motors in reverse. Out in the field I disarmed, turned the quad upside down and flicked the switch to enable Anti Turtle and rearmed, no motors armed. I tried again with no luck. I then approached the quad and turned it over and it armed and spun up the blades while turning it over and sliced my arm. I arm the quad buy holding the momentary and flicking the arm switch up, the throttle has to be at zero. Any ideas because at present it seems to be pretty dangerous.

Vidalcris, 2017-08-17:
I think you need to do that in cli : set small_angle = 180
Then you will be able to arm even upside down ;)

mikeller, 2017-08-17:
@AndyCWilson: Handling your quad while it is armed (on your transmitter) is a very dangerous practice.

AndyCWilson, 2017-08-17:
I know that Mikeller, what Im saying is that is should not have armed as the Transmitter was on the ground on its stand and it doesnt arm unless the momentary switch is held as well as the main arm switch. So why did it arm?

AndyCWilson, 2017-08-17:
Thanks for your useful information Vidalcris 👍

mikeller, 2017-08-17:
@AndyCWilson: The craft armed when you turned it around, since then all of the conditions for arming were met. The momentary switch only works on the transmitter, but once it's been held, the channel that is used for arming sends the armed signal. Not disarming before manipulating your craft is dangerous, and no amount of technical changes can fix this.

AndyCWilson, 2017-08-17:
Thanks for that info Mikeller. I will take note and double check it in future. Your explination makes things much clearer now.

beeb, 2017-09-19:
Since anti turtle mode is supposed to be used when the quad is upside down, shouldn't the firmware ignore `small_angle` when this mode is active?

mikeller, 2017-09-19:
@beeb: This has been fixed in #4166.