betaflight/betaflight #3875: 3.2.0 RC3 turning on blackbox logging was only working in cli

DEvil0000, 2017-08-17:
I was trying to turn on blackbox in 3.2.0 RC3 (F3 - eachine minicube) last evening. I configured the port, was looking for the feature to turn it on but there was none, turned it on in the blackbox page to "serial, 1kHz", clicked save & reboot but after that it showed again "no logging". I tried it three times then I opened the cli and turned it on there. Truning it on from the cli worked fine and after that it was shown correct in the blackbox config page and it was working correct, too.

DieHertz, 2017-08-17:
Which version of Configurator you had this on? There was a fix to configurator for the exact same behaviour.

DEvil0000, 2017-08-17:
3.2 was the version on a mac

mikeller, 2017-08-19:
@DEvil0000: 3.2 is a bit unspecific, as there are 3.2.0 / 3.2.1 ( / 3.2.2). Can you find out the exact version?

DEvil0000, 2017-08-19:
Sorry, it is 3.2.0

mikeller, 2017-08-19:
Ok, that's not the latest version. Try updating to 3.2.1 (uninstalling / reinstalling the app in Chrome will do this), and see if this is fixed in 3.2.1.

DEvil0000, 2017-10-24:
my new quad copter has build in flash - with this one it works at least. Can not test it with the old hardware anymore.

closed 2017-10-24 by DEvil0000