betaflight/betaflight #3876: Add OSD warning when battery not full

borisbstyle, 2017-08-17:
To warn the user that he is using a battery that is below ~16.5V a warning should be given to make the user aware that he is not using fresh battery.
This could be a feature that can be enabled and disabled. This is especially valuable in races where occassionally happens that user didn't exchange the battery

hydra, 2017-08-17:
👍

jaapp, 2017-08-18:
Great idea. Flying with a used pack also has another consequence: The capacity-remaining bar becomes misleading.

I've flow 5S packs until my video dropped out, because I was solely relying on the mAh-remaining indication. Maybe this capacity bar should blink or become unavailable on a less then full battery?

DieHertz, 2017-08-18:
I don't think so, be mindful of partially discharged batteries, keep track of how much you've used, have a safe estimate. Why disable it for everyone?

closed 2017-08-20 by martinbudden

jaapp, 2017-08-21:
I'm not (solely) suggesting disabling the capacity bar. Just raising an idea for improvement.

"be mindful" is a excellent advise. But preventing mistakes is even better :)

papadkostas, 2017-10-22:
But shouldnt we be able to skip that in case the battery is just a bit bellow the maximum?
I think there should be just a warning and not preventing arming the copter at all.

PR #3880 (2017-08-20): Add BATT NOT FULL warning to OSD
Shows `BATT NOT FULL` when the connected battery has an average cell voltage of less than `vbatfullcellvoltage` when the craft has yet to be armed.

Intended as a reminder to make sure pilots fly with a fresh battery.

Fixes #3876