betaflight/betaflight #3881: Current master causes insane yaw spin (Dangerous!)

Arakon, 2017-08-17:
Restoring a CLI dump from a week ago to the current master causes the yaw direction to be wrong (all my motors run reversed, and yaw_motors_reversed = ON). Board is Omnibus F3, happens on two seperate copters. Reverse box is also ticked in configurator. When throttling up, copter instantly starts spinning around yaw and shooting skywards.
Build is 27b146e274579d656339a7dd80e6a078369a1d13 (latest current master as of this post).

mikeller, 2017-08-17:
Good point, there is a `-` in https://github.com/betaflight/betaflight/pull/3873/files#diff-95e70d63cf1e155a276b2ad2a8e7a947L611 that was overlooked.

Arakon, 2017-08-17:
Fixed with your PR, thank you.

PR #3882 (2017-08-17): Fixed yaw direction after #3873.
Fixes #3881.
closed 2017-08-17 by mikeller