betaflight/betaflight #3891: OSD Setup in the field

Ralfde, 2017-08-18:
1.) OSD Setup in the field: I can not get into the OSD setup for adjustments in the field with Betaflight 3.2 RC3 or newer build
2.) After the flight I can go back from from the Post Flight Statistics to the flight display with pitch up and yaw left, but I can not get into the OSD setup menu.
2.) Omnibus F4 with MPU6000 and Graupner radio, travels adjusted to 125% in the radio, TAER1234 channel map.
3.) Add the used configuration and Betaflight 3.2 RC3 or build from today

Ralfde, 2017-08-18:
Sorry, I use the Omnibus F4 SD board with MPU6000.

mikeller, 2017-08-18:
@Ralfde: Have a look at your `min_check` and `max_check` settings, and verify that your TX can actually get below / above them.

Ralfde, 2017-08-18:
The problem was the throttle not in the middle. It was set to -125% with the motor stop switch together.
Thanks a lot. Somebody also helped me at rcgroups, otherwise I would not have found the problem.

So this issue is solved.

closed 2017-08-18 by Ralfde