betaflight/betaflight #3892: When "dshot reverse motors" is set in modes only motor 1 is active

grhookgr, 2017-08-18:
When "dshot reverse motors" is set in modes only motor 1 is active in any case even if the aux is not in the active range.
It seems as the fc does not send any command to other motors not able to access them even from motors tab
when the  "dshot reverse motors" is complete disable everything is back to normal

FC Matek F405 AIO OSD
Cicada 4in1 35a
Dshot150

grhookgr, 2017-08-18:
[LOG00035.BFL.zip](https://github.com/betaflight/betaflight/files/1234585/LOG00035.BFL.zip)

mikeller, 2017-08-19:
This sounds a lot like this problem: https://github.com/bitdump/BLHeli/issues/234#issuecomment-322019039

Try updating your ESCs to the latest firmware.

brycedjohnson, 2017-08-27:
What Esc did you have firmware?

mikeller, 2017-08-27:
@grhookgr: I revise my above statement, this sounds like it's solved by #3922. Can you update your firmware to a nightly build from ci.betaflight.tech to check if this fixes it?

RealPepon, 2017-08-30:
i Have the same issue , whit the 16.63 blheli firm just the motor 1 spin in reverse , the others motors continue spin in the normal direction

mikeller, 2017-08-30:
@RealPepon: Please read https://github.com/betaflight/betaflight/issues/3892#issuecomment-325170427

brycedjohnson, 2017-09-03:
I was still seeing issues with the reverse not taking place.  #4056 is a workaround that fixed the issues I was seeing.