betaflight/betaflight #3893: Yaw spin and climb after clipping gates - fix suggestion

ctzsnooze, 2017-08-18:
Hi guys

I've had multiple but infrequent instances of an insane power on yaw spin where the quad spins at more than 5000 degrees per second and climbs fast after clipping a gate or the ground on one side of the quad.

Oz posted a log clipping showing 2 motors full on - that's what it sounds like happens.  The motors stay full on even if throttle is cut to zero.  After about 3-4s the spin stops by itself (if you don't disarm) - by then the quad has climbed maybe 80m high.

I think that what is happening is that the quad passively spins at high speed, the gyros indicate an uncommanded yaw spin >2000 degrees/sec, and the yaw PI controller tells two motors to maximally oppose the spin; they go full throttle immediately, causing the climb.  It seems that they take quite a while to control the spin, but eventually they do.

Personally I'd prefer the quad to just drop, spinning, when this happens.  But it might be good to be able to regain control.  

I think it would slow down enough to eventually stop by itself even without any yaw thrust from the motors.  Maybe, maybe not.  But at least it wouldn't climb insanely and uncontrollably.  

It seems to me that a suitable solution could be to have a limit as follows.  When the actual yaw rate approaches some limit (maybe 500, maybe 1000 degrees/sec), the P and I values for yaw are multiplied by a factor that gets closet to zero the closer yaw gets to the limit.  If past the limit, yaw P and I are held at zero.  It would be nice if this factor has no effect alt low yaw rates but exponentially becomes more active quite close to the limit.

Note that the limit is based only on measured yaw gyro rate, not error or RC input.  

The effect of this would be that if the quad clipped a gate so as to cause a super fast yaw spin, the yaw PI controller would *not* attempt to control the spin.  Therefore the motors would not go full on in an uncontrolled climb.  It would remain responsive to throttle, and pitch and roll would still work (but not achieve much).  The pilot could then reduce throttle and let it crash controllably, or attempt to hover and wait for it to slow down, or disarm.  If they did wait, once the gyro measured yaw rate dropped back below the limit, the motors would attempt to control the ongoing yaw.  It could be that there should be hysteresis as to the re-start value; we would not want the quad would to suddenly lift a lot, and suddenly, at that point.

With that change, normal flight below the yaw gyro rate limit would be normal, the motors could go full on to generate yaw if required.

Some people like an option like all motors off until you have a forced crash and gyros all read 'zero' has little appeal.

spuder, 2017-08-20:
Betaflight version?
if 3.2, is Crash detection enabled?
Accelerometer enabled?

martinbudden, 2017-08-20:
@ctzsnooze , there is already a CLI variable `pidsum_limit_yaw`, default value is 400. Have you tried reducing it?

ctzsnooze, 2017-08-20:
@martinbudden - yes, I have tried lowering it, but then the maximum yaw responsiveness is affected all the time.  Sometimes when flying you really want a quick yaw responsiveness to a sudden yaw input (so you want lots of yaw PIDsum to drive that).  Even if you get a bit of an altitude increase, that's OK in these situations, because despite PIDsum going high, the motors are slow to respond.  So if I reduce PIDsum on yaw via the limit, yaw responsiveness can become sluggish.  Because yaw responses are so slow, the high PIDsum is really needed.  I've looked at logs where for very fast commanded spins yaw PIDsum is of the order of 400 are sustained while doing very fast 180 spins, so I don't want to reduce it.

What we really need is a limit - much lower, aka basically a number close to zero - which becomes active only when the quad is accidentally spinning super fast after clipping a gate.  In practical terms, this means a limit on yaw PIDsum that is only enabled when gyro rate for yaw exceeds a limiting value.  Or, better still, a limiting factor on yaw PIDsum that approaches zero exponentially as the gyro yaw value approaches the limiting value (so that it is a bit more gentle on and off).  

A value near 500 seems OK.  A solution like this would provide normal brisk  PID responsiveness when actual yaw rates are within the normal commanded range, and reduced yaw responsiveness (up to zero responsiveness) when the actual yaw rate is close to or past the limit.  

Hitting a gate and spinning faster than 500 would then result in zero response from the motors.  That would  be much safer than accelerating at high speed in some random direction.  If the quad slowed down enough that the yaw rate fell below the limit, yaw authority would return.

From memory the intent behind PIDsum limit was to prevent climbing / hopping upwards during fast yaw inputs, mostly for LOS pilots; pure hard flat yaw spins are unusual in racing, typically the only call for them is in fast 180 degree turns where there is a high quad angle, mostly a roll turn, and even if the yaw component of the turn is fast, a bit of lift is OK.  The default of 400 seems to be OK for the LOS person.  

It might be worth checking how hard the motors can/should be driven by a yaw PIDsum of 400?  Should it be possible for two to go full on at a sustained yaw PIDsum of 400?  They certainly seem to go full on after clipping a gate.

ctzsnooze, 2017-08-20:
@spuder no this was with 3.1.7 not 3.2, acc not on, and no crash detection configured.  I know it happens in 3.2 because a mate had it happen while flying 3.2 (without crash detection).

martinbudden, 2017-08-21:
@ctzsnooze , can you try PR https://github.com/betaflight/betaflight/pull/3909 - this sets a limit to the gyro yaw rate. Limit defaults to 1000, you may wish to reduce this.

inspector-71, 2017-08-21:
was it like around 2:50 in this vid and just after
[http://www.youtube.com/watch?v=EcbyCXpf_zI](url)
This is 3.1.7

etracer65, 2017-08-22:
I'm late to the party on this one, but I'm wondering if we're seeing an overflow situation coming from the yaw gyro.  I could definitely see clipping a gate at high speed causing a momentary > 200 degrees/sec uncommanded yaw.

https://github.com/cleanflight/cleanflight/issues/1818

ctzsnooze, 2017-08-22:
It could be an overflow issue of some kind.  TBH I don't know for sure what it is.  Inspector-71's video shows much the same event as what I had, but he gets control back a bit quicker.  Oz has also had similar issues.  So we know it happens...  just not sure if it is a 'normal' yaw PID response or not.

I asked Oz to try a *commanded* max yaw LOS - he obliged :-)  He can get his quad to 1200 deg/sec yaw in about 250ms with two motors going full power.  To maintain that yaw rate requires only 12% motor power.  It stops in about the same time frame with the opposite pair going full power. 

Given that degree of yaw authority, it does seem strange that clipping a gate can take 2-4 seconds of full anti-yaw to stop the spin.  I guess that fits with a yaw rate of 5400 deg/s in my case (I know my props don't generate quite as much yaw authority as Oz's, so it could well take me 4s to slow down from 5400 deg/s).

Martin, am I right in understanding that your solution means the FC won't perceive an incoming yaw value greater than the 'limit'?  ie even if the quad is yawing 5000 degree/s, then if I set the limit to 500, the FC will respond as if the spin was only 500 deg/s, and that should limit the motor drive.  But I am concerned that even if the quad was attempting to respond to 500 deg/s, it might still put the motors nearly full on, which wouldn't help.

My original proposal wasn't to limit gyro yaw, but to stop all yaw PID responsiveness during high rate uncommanded yaw spins, so the quad would not climb at all  Instead, while spinning, it could be dropped to the ground.  

I can see that if Oz wants to control 1200 deg/sec spins, then perhaps your limit suggestion might work well if it was calculated as the greater of abs (rc_command for yaw * 1.1) or the limit.  Hence the user could specify a very high rate yaw and the PIDs would respond right up to that rate, but for *uncommanded* spins the limit would be much lower.

Here is Oz's rig spinning, commanded, at 1200 deg/s:

[image]

etracer65, 2017-08-25:
I opened a new issue to discuss possible solutions for the gyro overflow.

https://github.com/betaflight/betaflight/issues/3959

ctzsnooze, 2017-08-26:
Good friend of mine just had a 100% loss of a Helix race quad with Kakute F4 ICM20689 board - clipped a gate in a race, insane spin upwards, by disarm time was very high, fell down on a busy road and got crushed, total writeoff.

richard-scott, 2017-09-13:
I'm getting an unexpected boost in height following a tug on yaw input on an SP Racing F3 Evo. Is this YSTTM? Remove yaw input and quad stops rising. 

Oh, and this isn't following a crash, but hovering and wagging on yaw stick to induce a 360 spin.

Here is my Diff, nothing much is enabled.
[SPRACINGF3EVO.txt](https://github.com/betaflight/betaflight/files/1300506/SPRACINGF3EVO.txt)

ctzsnooze, 2017-10-08:
https://youtu.be/SZWWlHXwnIg MPU6000

ctzsnooze, 2017-10-15:
Another episode of a yaw spin with an F4 board and MPU6000 yesterday on RC5, almost lost the quad.  A mate at the races last weekend says it's happening to his MPU6000 flight controller quads on release 3.2 but not on RC3.  So far no logs.  But yes something is definitely still not right.

fftunes, 2017-10-17:
> A mate at the races last weekend says it's happening to his MPU6000 flight controller quads on release 3.2 but not on RC3.

-_-

etracer65, 2017-10-17:
I haven't analyzed this yet, but I'm suspecting this commit introduced as part of 3.2.0 RC4.

https://github.com/betaflight/betaflight/commit/7fd3b143ca3b608d6c2a0724565c5d537bb1da99#diff-c5f9576172e273033324ba218524cc97

Of course a blackbox log of the event would be extremely helpful.  I only fly MPU6000 flight controllers and have never had a YSTTM event.

pulquero, 2020-09-14:
I lost a quad yesterday when I clipped the ground at a race and it yaw spinned off. Unfortunately, I didn't disarm quickly enough and I lost it beyond the tree-line. I would like to suggest a simple extension to yaw_spin_recovery. If no recovery is achieved within a configurable time limit, the flight controller should assume the pilot has lost control and enter fail-safe. Maybe also add an OSD message when yaw spin recovery is triggered.

closed 2020-12-25 by stale[bot]