betaflight/betaflight #3894: Cannot change through all LED colors using aux channel.

benwilliamson86, 2017-08-18:
When trying to change the LED colour via an AUX channel the LED only changes through a very limited spectrum (2 colours) and not all the colours available on the swatch. The Aux Channel ranges from 1000 to 2000. 

Using an F4 Omnibus on OBF4 on 3.2 RC3 - Aug 6 2017 13:33:15

https://youtu.be/1J17syNpsX0

benwilliamson86, 2017-08-19:
https://www.rcgroups.com/forums/showpost.php?p=38130264&postcount=51078

closed 2017-08-19 by benwilliamson86

mikeller, 2017-08-19:
@drclaes: Would you mind adding a page to the Betaflight wiki (https://github.com/betaflight/betaflight/wiki), explaining how to set up the new throttle LED?

DrClaes, 2017-08-19:
Hi!

I will do it! Just need to update everything at home!

/DrClaes

DrClaes, 2017-08-23:
Hi!

I wrote a page about the throttle functionality on LED strips. It’s on https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality … I have no experience editing wikis or even web pages, so I don’t know how to put the page in the right place.

Hope it looks OK.

/Claes Williamsson, Lycksele, Sweden

mikeller, 2017-08-24:
Thank you very much, I have added a link to it to the sidebar.