betaflight/betaflight #3896: DSHOT is not working on SPRacing F3 with Betaflight 3.2.0-RC3

jeremy-wendelken, 2017-08-18:
FC: SPRacing F3
ESCs: Emax Bullet 30A
BLHeli firmware: 16.67

DSHOT works fine in Betaflight 3.1.7. I have tried DSHOT150, DSHOT300, and DSHOT600

However, DSHOT of any kind does not work in 3.2.0-RC3

The ESCs continually beep (some kind of a warning or reset tone, I guess?), and the motors will spin up, but they continuously stutter (desync?).

Downgrading to 3.1.7 immediately fixes the problem, and DSHOT works without issue.

I would like to use DSHOT on 3.2.0 to make use of the new DSHOT reverse motor feature, but cannot with this issue!

DieHertz, 2017-08-18:
I flew with BLHeli 3.2 RC3 on SPRACINGF3 literally yesterday, DSHOT600, BLHS 16.67. Your issue is elsewhere.

jeremy-wendelken, 2017-08-18:
Why does downgrading to 3.1.7 resolve the issue then?

My flight controller is the one on the Wizard X220, that's definitely an SPRacing F3 right?

I don't have the small ground pad on the ESC connected, could that cause a problem?

jeremy-wendelken, 2017-08-18:
Multishot works completely fine.
Do I need to make any changes in BLHeli for DSHOT to work properly on 3.2.0-RC3?

jeremy-wendelken, 2017-08-18:
Here is a video of the issue occurring with DSHOT150 configured:
https://goo.gl/photos/7L21jyL87cPKnsJa6

jflyper, 2017-08-19:
@jeremy-wendelken Going back and forth between 3.1.7 and 3.2-RC does not solve your problem.
Do a full-chip erase flash or cli "defaults" then manually configure motors only without restoring from dumps from other releases, and see if they work.

closed 2017-08-19 by jflyper

richard-scott, 2017-08-21:
Did you remap your single outputs and remap Motor 4 onto Motor 5's output pin:

It's detailed in here [DSHOT ESC Protocol](https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol#spracingf3evo----must-move-motor-4-to-new-pin-assignment-cli--resource-motor-4-a06-then-solder-esc-for-motor-4-to-motor-output-5-fixes-dma-conflict-with-motor-outputs-2-and-4).

I have an SP Racing F3 Evo and that worked for me.

jflyper, 2017-08-21:
@richard-scott F3EVO has totally different timer mapping.

jeremy-wendelken, 2017-08-21:
The issue was having buzzer set to 1500 on the mode tab.

Without TX turned on this channel was defaulting to 1500 so of course the
buzzer was on!

In the motors tab, without quad armed, with ESCs powered up, I was spinning
the motors up while they were trying to play the buzzer tone!

This is not normally possible because it seems as though buzzer command is
not sent to ESCs when armed.

All sorted now. My bad!

Jeremy Wendelken
+64 27 297 9704

mikeller, 2017-08-21:
@jeremy-wendelken: That's a bug in the firmware indeed, the Dshot buzzer should not activate when the motors are spun up through the motors tab.

@brycedjohnson: Do you have time to look into fixing this?

brycedjohnson, 2017-08-21:
@mikeller I can take a look, but it might be a few days since I'm traveling

mikeller, 2017-08-22:
Too late, #3920. ;-P