betaflight/betaflight #3897: Dynamic filtering CPU load

thetobby, 2017-08-19:
I belive cycle time/load in configurator does not include dynamic filtering since PID loop is not running unless you arm your quadcopter.  Would this be correct understanding of how it operates or is my testing wrong/faulty?

I went from 
10% to 33%    on 2.63k/2.63k
30% to 715%﻿  on  4k/4k 
blackbox versus configurator with a SPracing F3 clone

If this is true, i suspect lots of user accidentally overload their CPU's.

DieHertz, 2017-08-19:
Gyro is sampled even when not armed, and dynamic filtering runs while not armed as well.

wind0r, 2017-08-19:
Only Motor output is disabled when the quad is not Armed. Also 75% isn't overloading the quad

DieHertz, 2017-08-19:
The original post said 715%, that's definitely not good, but 75% load is a good level.

thetobby, 2017-08-19:
Okay, so there must be another reason for this load. Thanks for clarification. Issue closed

closed 2017-08-19 by thetobby