betaflight/betaflight #3899: Omnibus F7 with SmartPort Telemetry not working in RC3

Fscked-In-The-Head, 2017-08-20:
I'm running an Omnibus F7 Pro from [Banggood](https://www.banggood.com/Omnibus-F7-Pro-Flight-Controller-Built-in-Dual-Gyro-AIO-OSD-Current-Sensor-and-LC-Power-Filter-p-1169518.html), and haven't been able to get SmartPort telemetry working for my X4RSB with BF 3.2 RC3. 

When attempting sensor discovery with a Taranis QX7, no remote sensors are found. The same receiver works perfectly on an SP Racing F3, also with 3.2 RC3 loaded, so the issue seems to be F7 specific.

I've tried UART 3 and 6, with and without inversion, and have tried enabling and disabling tlm_halfduplex, as that has allegedly shown issues with other telemetry methods. None of these steps have succeeded. Diff attached below.

\`\`\`
# diff

# version
# Betaflight / OMNIBUSF7 (OBF7) 3.2.0 Aug  6 2017 / 13:33:47 (981df1d54)

# name
name UBERQUAD

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 4 1950 2050
aux 1 1 0 1900 2100
aux 2 2 0 1400 1600
aux 3 13 2 900 1100
aux 4 26 4 1900 2100
aux 5 35 5 1900 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set blackbox_p_denom = 256
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 180
set yaw_motors_reversed = ON
set pid_process_denom = 1
set tlm_inverted = ON

# profile
profile 0

# rateprofile
rateprofile 0
\`\`\`

Fscked-In-The-Head, 2017-08-20:
It spontaneously began working after restarting the Taranis multiple times, if there's a software bug here it's with OpenTX.

closed 2017-08-20 by Fscked-In-The-Head

mikeller, 2017-08-20:
@Fscked-In-The-Head: Could be caused by a known bug in the RX firmware. Try updating the RX firmware to latest.