betaflight/betaflight #3900: Sbus vtx channel selection feature

Jahnkeanater, 2017-08-20:
So a few of the new vtxs coming out of china are using sbus channel 6 directly from the reciever to control the vtx channel. Can we add sbus vtx control to the osd and simulate a sbus rx signal with just channel 6 to control these vtxs with the osd? 

It would be nice to add this functionality before these vtxs start becoming more common. 

If anyone has one of these vtxs can they list what rx channel range results in what vtx frequency?

Jahnkeanater, 2017-08-20:
Can we also simplify the osd so it doesn't have multiple vtx channel sub menus? It would be nice if a single sub menu applied the settings across all 3 vtx protocols or just hide the features that are not enabled.

mikeller, 2017-08-20:
Not sure about this. This seems to be a not very well thought out approach (except maybe for fixed wing, if no flight controller is used). Supporting it will just mean that more manufacturers will jump onto the "SBUS VTX" bandwagon, instead of implementing a proper VTX configuration API.

Jahnkeanater, 2017-08-20:
Yea I really don't like it either but I understand not being able to use imersionsrc or tbs proprietary protocol. Or maybe there just too lazy to implement it.

DanNixon, 2017-08-20:
There already are cheap and nasty VTXes that support proper APIs (can remember off the top of my head if it was SmartAudio or IRC Tramp).

With regards to the OSD menu, it should be possible to group common operations (band/channel/frequency/power setting and status) but there is always going to be API specific menus given that the functionality exposed by each API is different.

mikeller, 2017-08-20:
The Matek VTX is an example of third party VTX that supports the IRC tramp protocol, and TBS have released a specification for the SmartAudio API, so that's also not strictly proprietary any more. My guess would be that the manufacturers want a piece of the 'remote controllable VTX' pie, but are too cheap to spend the money to develop something that is more than a lip service implementation.

jflyper, 2017-08-20:
> new vtxs coming out of china are using sbus channel 6 directly from the reciever to control the vtx channel.

Are these VTXs designed to peek at S.BUS stream?

Jahnkeanater, 2017-08-20:
Yes they tap into the s.bus signal. I think they only control channel and not band. I guess the fact that the FC has no way to determine what channel its tuned to is reason enough to not try to support it.

closed 2017-08-27 by jflyper