betaflight/betaflight #3901: BF 3.2.0 - beeper configuration -ON_USB faulty

TheIsotopes, 2017-08-20:
using latest jenkins build 1586 with options `beeper -ON_USB` ... it's works on usb but if usb unplugged and battery is connected the beeper is also turned off ... i investigated this and found the problem began with [jenkins build 1575](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1575/) ... @brycedjohnson please can you check #3510 and fix this issue

mikeller, 2017-08-20:
Sure it's #3510? It does not contain any changes to the beeper code. I would suspect #3831. Have you tried resetting your config to defaults, and then reapplying your settings?

TheIsotopes, 2017-08-20:
yes for sure it's #3510 ... tested jenkins build 1564 there #3831 merged and no issue ... tested also jenkins build 1574 and also no issue ... testing again jenkins build 1575 there #3510 was merged and the issue is back ... all tested after flashing with full flash erase and the same config

brycedjohnson, 2017-08-20:
I've be traveling and don't have any quads with beepers with so it might a
little harder for me to debug.

Are you running 4s on this quad?  Or something else?
Might be the changes with cell detection using filtered voltage? How many
cells does it detect after you plug in the battery?  What is the voltage
when plugging in there battery when this happens?

TheIsotopes, 2017-08-20:
my quads running with 3s only ... it is detecting 3 cells and the issue is happens with full battery

TheIsotopes, 2017-08-25:
@brycedjohnson any news how to fix this :((

mikeller, 2017-08-25:
@TheIsotopes: Patience, man. People are sacrificing their spare time to make Betaflight into what it is. Pestering them for updates will not increase their motivation.

brycedjohnson, 2017-08-27:
@thelsotopes I've been traveling and I won't be home for another week so I haven't had a chance to look at it.

TheIsotopes, 2017-09-10:
in #4107 the issue is described in more detail from @mikeller

TheIsotopes, 2017-09-11:
@mikeller @brycedjohnson tested now #4121 and it works perfect 👍 
hoping this will merged asap, closed this issue now

closed 2017-09-11 by TheIsotopes