betaflight/betaflight #3902: Implement over air time synch and use time for blackbox log create/modify timestamps

1337flite, 2017-08-20:
This is a feature request.

When I fly my quad and record blackbox logs the logs are recorded with the incorrect time.
I understand this is because the flight controller does not have an always on power supply and therefore could not track time over reboots if the time was set.

However I believe it is possible to send configuration commands to the FC over the air with radios such as the Tarranis and possibly other OpenTX radios and maybe even radios outside of OpenTX.

The Taranis and maybe  other radios do have a clock that keeps time.

_Is it possible to have feature where BF can be configured by over the air means to know and track the current time and use that as the timestamp for the creation date of blackbox logs?_
If it is possible can we please have this feature request triaged and added as feature request and maybe one day implemented.

I'm not sure if the BF codebase  actually contains the required date libraries/routines to translate ticks or seconds since clock initialisation or some other epoch event to human useful time so part of the feature request might be adding that

AirBreak69, 2017-08-20:
(not really needing this feature - just adding another idea for a time source)
If a GPS receiver is on bord the very best times source is available: The atomic GPS time is part of various GPS messages.
On GPS fix the time could be synchronized. But some additional information would be needed: target time zone, summer/winter time etc. Probably the blackbox explorer could handle this offline by the gathering information from the host computer.

fbohmann, 2017-08-20:
this is a duplicate of #2729

closed 2017-08-20 by DieHertz

1337flite, 2017-08-20:
I did a search of the issues for this repo but didn't find #2729 - apologies.