betaflight/betaflight #3904: Voltage refresh rate to fast (OmnibusF4)

DerDominik, 2017-08-20:
Last days I tested BF-3.2RC3 on my OmnibusF4 (don't know if other FCs are affected) and it seems, that the refresh rate for voltage (maybe also current) was increased. In my mind it is to fast for checking the voltage.

For example 2 videos.
BF 3.1.7 -> https://youtu.be/7pSAi3_IQrE
BF 3.2.0RC3 -> https://youtu.be/5JVS4xYnOhw

I think it makes more sense to decrease the refresh rate or take x measurement points a minute and display the average value from this points, like in old BF3.1.7.

mikeller, 2017-08-21:
@brycedjohnson: Could this be the result of your changes to VBat filtering in #3510.

brycedjohnson, 2017-08-21:
These changes weren't in rc3.  #3510 increases the filtering on on the filtered vbat so it should actually fix this issue.

DerDominik, 2017-08-21:
OK, I'll test an actual build of BF the next days.
Thanks for your help :-)

DerDominik, 2017-08-23:
Yesterday tested an actual build and its working like old versions.
Thanks for that great work!! :-)

So I'll close this isue

closed 2017-08-23 by DerDominik