betaflight/betaflight #3905: Omnibus F4 V2 Kompass anschließen

gerse70, 2017-08-20:
Hallo,
ich habe eine Omnibus F4 V2 und möchte einen Ublog MGN8 anschließen.
Das GPS habe ich an J5 angeschlossen und über UART1 freigeschaltet, es Funktioniert!!!

Wo klemme ich den Magnetometer an?
Hatte ihn an J10 an SDA undSCL angeschlossen, geht nicht!
Muss man noch was programmieren?

Beta 3.2.1 aufgespielt

mikeller, 2017-08-20:
Ja, Du musst den I2C bus erst noch konfigurieren, weil die Pins fuer I2C **und** UART3 genutzt werden:

- UART3 darf nix konfiguriert haben;
- in CLI:
  `resource i2c_scl b10`
  `resource i2c_sda b11`

Uebrigens, die meisten Benuetzer hier sprechen englisch, von dem her wirst Du schneller eine Antwort kriegen wenn Du in englisch fragst.

jflyper, 2017-08-20:
@gerse70 

You sound like you have a OMNIBUS F4 Pro v2, not OMNIBUS F4 V2.

- GPS on UART1 (J5) should work. Review your GPS settings in configuration tab.

- To enable I2C instead of UART3 on J10, you will be using the following CLI command. 
```
resource i2c_scl 2 b10
resource i2c_sda 2 b11
```
Note that only HMC5883/HMC5893 is supported as an external compass device. If your compass is one of these, it will be automatically detected.

For further assistance, please redirect your questions to RCG thread, as this is not a place for user support. Thank you in advance.

Ref:
Omnibus f4 pro
https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro

closed 2017-08-20 by jflyper