betaflight/betaflight #3906: anyFC Flight Controller/ When Uart is enabled, motor # 3 stops spinning when armed.

jlbiarlee, 2017-08-21:
I have the anyFC with 8 open Uarts. I have enabled 2 of them, but one is for my Rx and the other is for my Smart Port. Whenever I enabled any of the others my #3 motor stops working. I am wondering if I am doing something wrong, or possibly somebody else has had this same issue. I have searched all over the internet but their is a lack of info protaining the F7.
 If you enable a Uart that I need for RX and TX the #3 motor stops spinning when armed. When I disable the Uart all motor arm and spin correctly and it flies great.
 I have a anyFC F7 Flight Controller, Frysky RX, 4 in 1 ESC, and basic 2204 motors.
 I have flashed the new 3.2 Betaflight and also flashed 3.17 Betaflight, neither has changed the outcome.

mikeller, 2017-08-21:
Sounds like a resource conflict.

Check the output of `resource list` before enabling an UART, then check again after, and make sure that the UART you have enabled has not taken over the pin or DMA for motor 3. If this doesn't work for any of the UARTs, you might have to move motor 3 to a different pin.

jlbiarlee, 2017-08-21:
Well I have to say, thanks for quick response. I will check that out. I was thinking it was something along those circumstances, but I am not to familiar with using the CLI tab, but I will check it out. Thanks again.

mikeller, 2017-08-21:
There's a good primer for resource remapping in the CLI: https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping

jlbiarlee, 2017-08-21:
wow, once again, thanks for the fast response...I have never ever asked a question on the "internet" I just read and hope to see someone else that might be having the same problem, but this is great, I'm gonna plug in right now and see what and where whats mapped out, then enable uart, then look again at resource map and compare correct?

jlbiarlee, 2017-08-21:
what does DMA mean? I think Im getting somewhere...it wasn't motor 3 but it was 2...my mistake, but when any uart is enabled the motor 2 disappears, and the uart TX takes over...should I move motor to different pad?

mikeller, 2017-08-21:
Direct Memory Access. With respect to your problem, it is just a resource that can be used by a motor pin (if Dshot is used) or an UART, but not shared.

And yes, move the motor to a different pad, or find an UART that uses different resources.

jlbiarlee, 2017-08-21:
Thanks man,
Ill check it out

jflyper, 2017-08-21:
This is the side effect having UART TX DMA on by default, tried to fix it by #3885 but reverted by #3887 because the UART driver does not support non-DMA TX.

@sambas We need non-DMA TX for F7.

jlbiarlee, 2017-08-26:
so is there fix for this? or is it in the making?
I'm new to all this, are you guys the ones that actually write the code for the FC's.... etc?

jflyper, 2017-08-26:
@jlbiarlee Yes, we are writing codes, but we each have specialities, so can not fix arbitral code. The use of DMA for UART TX is hard coded deep inside the serial driver code for F7, and it can not be turned off at the moment.

This means there is no way to stop UART TX from using the DMA resource. This leaves you three options.
Option 1. Give up using Dshot.
There is still a chance that DMA resource requests may collide among UART TX and other uses (such as ADC). You still have to watch DMA resource usages as described below.

Option 2. Carefully decide which UART to use so that DMA resource requests do not collide. Note that RX side does not use DMA, so RX-only uses such as SBUS has no restriction as far as the DMA resource is concerned.
Current DMA resource request are as follow.
For motors,
Motor 1 DMA1 Stream 7
Motor 2 DMA1 Stream 1
Motor 3 DMA1 Stream 4
Motor 4 DMA1 Stream 6

For UART TX,
UART1_TX DMA2 Stream 5
UART2_TX DMA1 Stream 6 // Collide with Motor 4
UART3_TX DMA1 Stream 1 // Collide with Motor 2
UART4_TX DMA1 Stream 2
UART5_TX DMA1 Stream 7 // Collide with Motor 1
UART6_TX DMA2 Stream 6
UART7_TX DMA1 Stream 1 // Collide with Motor 2
UART8_TX DMA1 Stream 0

This means that you can only use UART1, UART4, UART6 and UART8 if you want to use TX side and Dshot at the same time.

Option 3.
Remap motors so they don't collide DMA resource requests.
This requires relatively deep knowledge of timers and DMA mappings.

Hope this helps.

jlbiarlee, 2017-08-26:
wow, thanks for breaking that down for me. You really nailed it. Sorry for so many post in different commits- I am trying to figure out how Github works, and how to understand this literature... again, @jflyper thanks a lot for going out of your way and laying out the options. Do you think that the issue can be fixed one day in the future? I'm in no hurry, I'm just curious... I wonder why the maker of this flight controller would sell an item that doesn't work as expected. No worries... I will try my problem solving. Thanks again! 
Judson 
Central FL

jlbiarlee, 2017-08-26:
I am guess that SmartAudio runs on a different kind of Tx? I have that working with my Lua Script on OpenTx...but I have tried hooking up GPS, Bluetooth, and Vtx ... there is still a lot of terminology I need to learn when it comes to pinouts and how to read. I have a general idea, and have been wiring up flight controllers by soldering directly to FC, but it seems like the things I get from China are a little bit on the "screwed up" side of things...

jflyper, 2017-08-27:
@jlbiarlee Devs are constantly and consistently looking into issues and new features, and this DMA resource collision would certainly be fixed relatively soon, as it affects most of the F7 targets. I understand that many of ANYFC users did buy one because of the rich connectivity options, and it's frustrating not be able to execute their original plans.

closed 2018-04-21 by DieHertz