betaflight/betaflight #3910: Fortini F4 on 32KHz gyro update

madwasp, 2017-08-21:
I've installed the Betaflight 3.2 RC3 in my Fortini F4 FC and everything is working almost perfectly.
The only thing I've noticed is that, when I enable gyro update to 32KHz, when I arm the quad it just gets crazy and starts to takeoff immediately by itself making some noise vibrations.

After disable the 32kHz gyro update on betaflight everything works perfectly on lower kHz updates.
In terms of hardware it's supposed to work properly on 32KHz gyro.

Going to make more tests meanwhile.

mikeller, 2017-08-21:
'supposed to work' is a bit of a stretch. The 32kHz mode on the gyro is not supported for production use by the gyro manufacturers, since it bypasses all gyro side filtering. In reality, this causes a lot of gyros (as in, individual chips from the same series) to be very noisy, to the point of being unusable. A different chip of the same series can work fine at 32kHz. So maybe try another board and see if it works on 32kHz.

Aileron8, 2017-08-24:
Mine was the same until I removed the nylon nuts holding the FC down. So now my Fourtini is floating on gummies. Also make extra sure your wires are small and not transmitting any vibration. 32k works, you just need to eliminate all sources of vibration.

closed 2017-08-30 by martinbudden