betaflight/betaflight #3911: Add Custom TPA curve adjustments

oyvinla, 2017-08-21:
Would it be possible to implement custom TPA curve adjustments similar to kiss or raceflight?

tigert, 2017-08-22:
I think a great start would be to collect some examples/desired setups in the wiki and try to populate them with values that work?

fftunes, 2017-10-22:
Imho it would make sense. Wonder why nobody cares about it.

To make it even more useful - how would it be if it could be abused to boost PIDs beyond a scale of 1 in certain throttle areas...

Jonitsvk, 2018-07-13:
Yes that would be really helpful. Please consider implementing this feature.

Ralfde, 2019-03-12:
BF 4.0.0 RC1 does work really great, but even with perfect filtering the feature mainly the  P TPA curve is still missing to setup the Quad perfectly. Second importand would be a D TPA curve and less important the I TPA curve.

mikeller, 2019-03-12:
@Ralfde: Feel free to open a pull request implementing this for 4.1.

closed 2026-09-26 by blckmn