betaflight/betaflight #3912: MATEK FC F405 AIO yaw twitch and drift

Preathor, 2017-08-21:
I'm using Matek FC F405 AIO and I'm experiencing some yaw twitches and uncontrolled yaw spins. 
After reverting back to 3.1.7, the problem seems to go away. I've tried this at 8k/8k, 8k/4k, dshot600, antigravity at default values and airmode turned on. Everything with different standoffs, softmounting motors and softmounting pdb. Since, reverting back to older version, I'm just thinking (but not claiming) that it must be something in new target version. I haven't done anything special with configuration either... everything is basically just "stock". I've also turned on active notch filter and turned off all filters and used pt1

My particular build is: 
- armattan chameleon 5''
- armatan 30amp esc
- emax 2305s red bottoms
- matek fc f405 aio
- frsky xsr
- tbs unify race 

I don't have any black box data, but this youtube example shows what I mean (not my video, but fairly similar issue): https://youtu.be/tSvOhG7PK5M?t=63
I've also experienced slow drifts in yaw direction, when I had to compensate with my sticks to go forward normally. 

Problem is visible on video at 1:06 and I hope I'm not violating any licences by posting this video here. 

Best regards, 
Link

tigert, 2017-08-22:
This is a shot in the dark, but I had this also on 3.2RC3 on a "ISHAPEDF3" (A Banggood clone of "Lulz F3" board they sold a while ago).

It would also show as yaw twitches (short movement in the 3D quad preview in the initial configurator screen).

I am not sure what it was, but I reflashed it wiht 3.2 and 3.2.1 configurator and rebuilt my config from scratch and it seems to have gone away now. I assumed it might have been a damaged gyro chip, but seems like it is stable now.

xlla, 2017-08-23:
I am using that board too, matek f405 aio with bf 3.1.7 firmware.
It will yaw spin and yaw twitch when I quick raise throttle, or do a quick yaw spin.
It fly smooth on hover.

Another quad with same device fly fine. I think that was caused by a bad gyro on FC.

Preathor, 2017-08-24:
What's interesting to me is that in my case, it works fine on 3.1.7, smooth flight, no problems at all whatsoever, but when I upgraded it to 3.2 it started to show problems. When I reverted back... worked okay again.

I've been following the FB page for this particular board, and I'll try some of their "hacks"... moving from uart2 to uart5 then report back how it goes (next week)

rossonado, 2017-08-26:
I have the same issue. New build 3.2. Quad is OK, but if punch out and throttle back, quad drifts into a spin on its own. Same at most points when inverte.  When flying normally it's relatively OK. I've upped timing, raised, p, I and antigravity and messed with filters and nothing changes It. I've changed escs with no difference. It feels like airmode is not on. System is clean, cool motors etc. Rebuilt quad to add more softmounting. Will try 3.17 and see if it helps. Seems like this is an amazing fc if you're lucky and awful if not. Regroups is full of it.

robert-hoffmann, 2017-08-28:
I had some slight yaw twitches when i put in a (very) tired battery, put in ta new one and twitches went away ...probably not related though

Peter-JanGootzen, 2017-08-28:
I have flown 15 batteries with my new build with the Matek F405-AIO. I have had zero issues **whilst running 8k/8k**. No twitching and very little noise. I have flown with 3.1.7 and 3.2 RC4, both been great.

But, when I try to run 16k/16k my motors don't arm at the same time, the motors also sound way different and when trying to take off it just tumbles into the ground. Very weird stuff. I think this is some weird configuration issue, I am running DShot600. This issue is happening on both 3.1.7 and 3.2 RC4.
If someone wants to help me trouble shoot, I can post all my BLHeli and BF relevant settings.

@Preathor This sounds a lot like you do not have airmode turned on. If you are currently using airmode as a feature, you might wanna try using it as a mode. There have been people who have been having problems with that before.
https://www.youtube.com/watch?v=6GcsRXr9gzQ
https://www.youtube.com/watch?v=EQ-qxBsMl4w&t=333s

Preathor, 2017-08-29:
I have airmode turned on correctly. I don't think this is the problem. 
@Peter-JanGootzen, how did you connect your RX and which one are you using? Did you use diagrams from Matek site? 

I've been seeing more and more people posting videos of this twitching on Matek FB page. It's a shame, because it's a nice board, and so many people reporting problems with it.

Peter-JanGootzen, 2017-08-29:
@Preathor
I am using the Flysky X6B and it is connected to uart2.
Yes, it really is a shame. This is an amazing FC, when it works...

TimothyGold, 2017-08-30:
I also have used the FC in a new build and have struggled with this form the beginning.  No matter what I try I could not get the yaw issue under control.  I originally built with the black included vibration dampers and to be honest I never saw any vibration issues in Black Box that seemed too severe.  In fact Pitch and Roll was very good.  Yaw seemed to be worse than pitch and roll but didn't look worse than other examples i have seen.  I eventually lost control of the copter and could not recover and now have to rebuild.  In my rebuild I was going to try softer isolation to see if something would change.  I am starting to wonder if I just need to settle on a different FC maybe using a more reliable IMU not the ICM20602 that seems to be plagued with issues in 3.2.

s700ky, 2017-08-30:
Put my 405 AIO into a Crusader 175. 
Violent yaw twitches. 
All filters on and BiQuad set and it calms down a little, but still unflyable. 
Reflashed Board and ESC’s still the same. 
Tried all different ESC settings still the same. 
Running 16K 16K still the same. 
Need to try 8K 8K as suggested above. 
Do not have this problem on the Matek board with separate PDB.

s700ky, 2017-08-30:
Also even though I installed the board with the arrow facing forward I had to yaw the board 270 to get its orientation right.

rossonado, 2017-08-30:
You always have to do that. It's because it's the same firmware as the normal board.

Preathor, 2017-08-31:
So.. I did some testing. I don't know if it's my particular board that just might be bad, or is this a general thing for this board. I'm still reading about same problem on FB page. Some have managed to solve it (with some sort of luck and leprechaun magic), but I'm not actualy sure.... 

I've replaced the reciever with X4R-sb (instead of XSR)... resoldered wires on it so that i have uninverted sbus and smartport (telemetry) using Project blue falcon and Joshua Bardwells instructions. That gave me the option to move these two things to a different uart. I've disabled soft serial, disabled blackbox, i've been using 8k/8k, I've enabled airmode as a feature (in configuration tab, so that it's always on), I've disabled accelerometer, tried different tpa values and antigravity values... pt1 without any  notch filters, enabled new active notch filter, tried different soft mounts... I've replaced my emax motors with mr. steele silk motors... the white ones... (because emax ones can be a little noisy) did new pid tunnings... aaaaaand... 

... same deal :( 

Thing flies okay for half a pack... but one I do a couple of hard manevers, it starts to drift to the right... I always have to compensate with my yaw input... this does for about 30 seconds (slow drift)... then it "let's go", makes a twitch... and it's back to semi normal. Until i do another manever.. split-s for example...then it's starts to drift again.  

I don't have any black box logs because I've been having some troubles with sd-cards... and I don't have any high-res/low-res footage of it. I'm not putting my gopro on that thing. 

I'll be reverting to BF 3.1.7 tomorrov and see what happens again. 

I don't mind staying on 3.1.7, but I was liking that active notch filter thing... Shame really. Might aswell switch back to my Flip32, or maybe buy Kakute... If weather won't allow it, I'll start doing Matek diffs on 3.1.7 version with 3.2 :D

Anywho... still best kuddos to BF developers. Keep on doing the great work.

Best regards, 
Link

robert-hoffmann, 2017-09-01:
Still no twitches here.

Did have motors a bit warm, so bumped down gyro/dterm soft lpf filtes by 10 points each. Now they are barely warm.

Also soft mounted the motors (fc was already soft mounted), and seems to fly much smoother. I already went with PT1 on gyro when i bumped the lpf down, and tomorrow i'll see if i can go back to PT1 on dterm too.

Still running 8k/4k gyro/loop on dshot 600

1337flite, 2017-09-04:
Works for me.

My config is nearly the same as Preathor  other than  Wraith 32 35A ESCs and Red bottom 2205S and I have no problems.
I have 220uf 25 volt caps on my ESCs and the Matek provided cap on the battery leads and its flys beautifully on 3.2 and has been since RC2.

I have a similar set up but with the Matek F4 and PDB and that flies just as well on 3.2 since before the release candidates. That's a 6" Cham and has different motors (BoltRC 2207 2450s).

Motors on both are very soft mounted and FC uses matek supplied rubber mounts.

robert-hoffmann, 2017-09-05:
Check this out guys
https://youtu.be/uINN6AXVhFE

I didn't have any problems to begin with... But settling motor pwm to 48khz (if your on blheli32) does bring the global noise envelope down.

Now on zero notches (except dynamic) gyro/pid 4k/8k with both on PT1, and motors barely warm. Motors/FC soft mounted, and 470 if cap on battery.

Gonna try 8k/8k and dshot 1200 and see if I can bump the lpf back up some soon.

chriswils, 2017-09-05:
Here is my blacbox https://drive.google.com/open?id=0B9Lnsj9U0mK8TkYyRmRpUEJnYmM with bf 3.1.7. 

A lot of twitching and sometimes yaw spin. 8k/4k. 3% CPU load in idle.
RCX 2206 2450kV
Racerstar 30A LiteS
DSHOT 600

I have the exact same setup, but with Emax RS2205 2300kV, and no such problems. Also, I saw that my CPU load spiked to 100% when acc was enabled. `tasks` showed 32%. 
When going in to turns, the PID's got really aggressive. This is default PID's.

Hustleflyer, 2017-10-04:
I have gone through 3 different builds all with Matek F405 AIOs, and two out of three experienced this exact problem, mine was much worse with one I was debugging and troubleshooted all the way to the Gyro itself after rebuilding same machine three times, an Arris X220 with two different sets of motors and then different ESCs just to be extra double sure.  I tried to band aid with a rubicon capacitor and it did seem to help a little, only other solution is a one capacitor per ESC. The other two had less noticeable gyro issues and more signal interference only when throttle was applied, signal noise would appear and I could not manage to make it disapper, no matter how I wired the power to cam   Did not have to take logs to confirm the basic fact the gyro on these boards, though fast, has many noise issues coming from nearby emi -electro-mag interference.

jalao, 2017-10-20:
Please please help. . Is there going to be a solution in sight for the agreesive Yaw twitch of this board? If I slowly apply throttle while making turns, the quad flies well, but the moment you make aggressive turns it will twitch hard and it some instances a complete spiral crash. I have soft mounted the board. Soft mounted the motors in two places. Directly under the motor and at the screws that hold the motors. I have used the capacitor provided with the board. I am running the Matek AIO board, new f60 tiger motors , wraith32 escS. I am running 8k/8k. Running airmode, dynamic notch with pt1 enabled.

JMcDee, 2017-10-22:
I'm having the same issue, running 3.2.1 firmware. Violent yaw twitches, roll and pitch are perfect.

Frame :       RED RD-025 TALON SVX 220 low side plates 
FC:              Matek F405 AIO
ESC:           Racerstar Tattoo_S 35A 4in1 2-4S STM32F051/ARM Blheli_32 Dshot1200 Ready Dual BEC FPV Racing ESC
VTX:           TBS Unify pro 5G8 V2 (SMA)
RC RX:        TBS Crossfire Micro RX V2
MOTORS:   EFAW 2407R 2500KV
CAM:          Runcam Swift 2

rossonado, 2017-10-22:
It's the board. In the newer ones, it comes with four capacitors. It's clearly got voltage issues in quite a few of them.

DimaVIII, 2017-10-23:
Hey Guys,
To solve this problem you need to place a low ESR Cap on each ESC.
They include now 4 Caps in the package inserted of one of you are lucky and get the new batch.
Check out this: https://www.rcgroups.com/forums/showthread.php?2912273-Matek-Flight-Controller-F405-AIO/page33

Fuzzball27, 2017-11-28:
Is this issue solved?
Is this fixed by placing caps on the ESCs, or by reverting firmware?

jalao, 2017-11-28:
Caps on all ESC and on main battery led did the trick.

martinbudden, 2017-11-28:
Closing, it's a hardware issue.

closed 2017-11-28 by martinbudden