betaflight/betaflight #3916: DSHOT Full throttle after arm (maybe related with DSHOT beeper)

mjedlinski, 2017-08-22:
My hardware:
Betaflight F3
Kiss24 ESC
F40 II 2400KV

Software:
Betaflight 3.2 RC3

Issue:
I made many flights with BF 3.2 RC1 and RC3.
This issue happend two times. I have video and blackbox:
First time BF 3.2 RC1 https://www.youtube.com/edit?o=U&video_id=Yho8zM4fJiI
Second time BF 3.2 RC3 https://www.youtube.com/edit?o=U&video_id=C37yz9mCjag

After power on initialization is ok. When I arm my quad one of motors go full throttle.
After dissarm 3 of 4 motors stop working (dissarm), but the one that goes full throttle keep going full throttle until pull out battery. I fly houndred times on 3.1.8 with this same setup and everything was ok.

It may be releted to dshot beeper because my arming sequence is:
DISSARM -> BEEPER -> BEEPER OFF and ARM

DanNixon, 2017-08-22:
Please use public facing YouTube links.

https://www.youtube.com/watch?v=Yho8zM4fJiI
https://www.youtube.com/watch?v=C37yz9mCjag

mikeller, 2017-08-22:
@mjedlinski: What firmware version do you have on your ESCs? (If you use ESC sensor, a `dshotprog 255 6` will give you the answer.)

mjedlinski, 2017-08-22:
I use KISS24A_110e_RC9

closed 2018-01-01 by mjedlinski

RipperDrone, 2018-01-01:
is it fixed? Or just closed? Pls leave a short comment how you fixed it (in case you have) so other users can learn from your reported issue... thank you.