betaflight/betaflight #3917: Blackbox logging - logs unreadable/crashing BB explorer

zytra, 2017-08-22:
Hi,
I have a problem reading the BB log generated by BF 3.2.0 RC3.
I am running BrainFPV RE1, configurator 3.2.1.

Logs are generated just fine. I initially thought my log was a little too big (11MB 4mn flight @ 2KHz logging) but repeated the process for only a few seconds of flight and same results.

BB Explorer hangs, 1x CPU core running @ 100% - left it for 20 minutes and nothing. 

Is there a special version of BB explorer needed?

thanks

JonnyPhenomenon, 2017-08-24:
I had this problem too.  I **think** I fixed it by removing the blackbox extention under Chrome://extensions and then re-adding it... although, it might have been fixed when I re-flashed to a more recent build of 3.2...  

Are you writing the logs to flash? or does that board have an SD card?

JonnyPhenomenon, 2017-08-24:
what version of the blackbox explorer are you using?  the most recent version is 2.5.13
Updated: August 10, 2017 - I know its two weeks old now, but I believe mine did not auto update.

zytra, 2017-08-24:
I tried both 2.5.13 (latest) and 2.5.12 (latest official) on 2 machines and even sent the log to a friend who had the same issue with it.

I brought the issue up to the BrainFPV guys who compile BF builds separately from the main repo; just in case it's a board specific issue.

that board has 16MB of flash memory built-into it. 

thank you!
Cheers

JonnyPhenomenon, 2017-08-24:
have you tried logging at 1khz instead of 2?
could this be a blackbox explorer issue? like https://github.com/betaflight/blackbox-log-viewer/issues/77
or https://github.com/betaflight/blackbox-log-viewer/issues/67
( https://github.com/betaflight/blackbox-log-viewer/issues )

zytra, 2017-08-24:
That sounds very much like it might!!!
I will give this a shot today, thank you as I was not aware of this issue.

I will report back later today.
Cheers

zytra, 2017-08-25:
Quick update, I didn't get the chance to test that tomorrow but should be able to tomorrow.
thanks again

mikeller, 2017-08-25:
Yes, I think there is a problem with the blackbox explorer. Or rather, at today's logging rates, the log files seem to get to a size that is too big for the explorer to handle.

However, this is not a firmware, but a blackbox explorer issue, can you please close this and reopen it (with a link to this issue) here: https://github.com/betaflight/blackbox-log-viewer

zytra, 2017-08-25:
I did logs as short as 10 seconds @ 2 KHz (less than 500 KB) and still had the same issue.
I opened logs upwards of 6MB on kiss and that same version of BB explorer had no problem opening it.

I can also open logs made with 3.1.7, on the same FC with no problem either.

I will keep this open for now.

zytra, 2017-08-25:
I just did a quick hover in my garage, logging @ 1KHz. 320 KB file.

[BTFL_BLACKBOX_LOG_20170824_213008.BBL.zip](https://github.com/betaflight/betaflight/files/1250843/BTFL_BLACKBOX_LOG_20170824_213008.BBL.zip)

Still can't open it with BB explorer.

mikeller, 2017-08-26:
Not a firmware issue.

closed 2017-08-26 by mikeller

sepal, 2017-09-16:
I have encountered the same issue. Logged several short hover flights with 1.5khz and the file has now a size of 3.4MB. Apperently the blackbox log viewer hangs on the loop in the countIntentionallySkippedFrames() function in flightlog_parser.js:1108. I don't know enough about betaflight and the blackbox features works, so it's kind of hard to debug the problem. 
FC was a kakute F4.
[blackbox_sepal.zip](https://github.com/betaflight/betaflight/files/1308598/blackbox_sepal.zip)

sepal, 2017-09-16:
Commenting out the call to the function mentioned above helps. Just go to flightlog_parser.js:1145 and comment out that line. I was able to open my log file with that hack, I have no idea what side effects that will have though.

sepal, 2017-09-16:
Hmm, should probably move my comments to betaflight/blackbox-log-viewer, right?

mikeller, 2017-09-16:
@sepal: Yes please.