betaflight/betaflight #3919: RPM BLheli-32

Gaz-B, 2017-08-22:
# If your issue looks like hardware fault or a configuration problem this may be is the wrong place to post to.
## Please consider to using [RCGroups](rcgroups.com/forums/showthread.php?t=2464844), [Slack](slack.betaflight.tech) or any other forum before posting here or even better ask the manufacturer of the hardware you are using.

## If you post here please follow these steps
1.) Describe your problem
2.) Include ways to reproduce the problem
2.) Provide information about your flightcontroller and other components
3.) Add the used configuration and firmware version
4.) Create a diff and post it here in a code block (Simply put \`\`\` at the start and end of the diff block)
5.) Remove this Text :)

Gaz-B, 2017-08-22:
Will there be a way to configure the osd and logging to reflect the correct motor rpm with BLheli-32 telemetry

mikeller, 2017-08-22:
@Gaz-B: Yes, we could be adding a conversion factor from eRPM to RPM when we are adding the RPM to OSD / telemetry.

mikeller, 2018-06-03:
@Gaz-B: This has been implemented in #5766. I think this can be closed.

closed 2018-07-07 by mikeller