betaflight/betaflight #3926: FailSafe Betaflight 3.2 RC3

mStrangers, 2017-08-22:
i saw a strange behavior in 3.2 not present in 3.1.7 and before.
if i activate fail-safe from a switch when the quad is armed , the quad disarm , throttle set to 885 all is normal.
but in betaflight 3.1.7 and before if  i disable the fail-safe switch will the arm switch is in arm position nothing append , i have to disarm the arm switch and rearm it to arm the quad it's the normal thing to do
in betaflight 3.2 RC3 if i toggle the fail-safe switch the quad disarm ,throttle to 885 all is ok, but if i toggle the fail-safe switch will the arm switch is in arm position the quad arm and this is not a good thing.
tested on different board (omibus f4,matek f405)

DanNixon, 2017-08-22:
That is normal and intentional.
It is a precaution to prevent accidental arming.

Why are you operating the failsafe mode on a switch?
That is intended to be used with RX failsafe to indicate failsafe conditions to the FC.

mStrangers, 2017-08-22:
the 3.1.7 is normal and prevent accidental arming i use it for this , but in 3.2 it didn't work as it.

DanNixon, 2017-08-22:
If you want a switch to prevent accidental arming then you should use PREARM.
Doing this with the FAILSAFE mode is a misuse of the mode.

mStrangers, 2017-08-22:
ok i know that but no other method before 3.2. in 3.2 if it work like this, if the quad lost radio and enter in failsafe mode and the radio reappear it rearm the quad ,and i thing isn't the right thing to do ,even if i misuse the failsafe switch.

DanNixon, 2017-08-22:
That isn't how it works.

mStrangers, 2017-08-22:
so if i use failsafe switch to indicate failsafe to the fc from rx remote lost , and the rx retrieve link to the remote ,it toggle the failsafe switch and the quad rearm ? no ?
I'll test the 2 scenary tomorrow

DanNixon, 2017-08-22:
What RX are you using?

mStrangers, 2017-08-22:
i use Frsky Xsr and i use no signal on failsafe ,it isn't the purpose, i can use any rx that exist and someone didn't have nosignal (like flysky) or my misusing of the failsafe switch.
the purpose is why 3.1.7 and 3.2 didn't work the same when it leave the failsafe mode activated by a switch.

DanNixon, 2017-08-22:
> i use Frsky Xsr

Right, in that case leave the FAILSAFE switch alone, you don't need it.
If you want to add additional safety then use the PREARM mode on an AUX switch.

The behavior is different in 3.2 because it was unsafe in 3.1.7.
If you enter failsafe stage 2 the craft should not arm again unless commanded to by the pilot.

I'm not at all sure what you thing the problem is.

mStrangers, 2017-08-22:
if i enter failsafe stage 2 with a switch and the arm switch is on , if i leave this mode the quad didn't arm in 3.1.7 but it arm in 3.2

libwtf, 2017-08-22:
i agree with @mStrangers , after a failsafe (switch or otherwise - i did, as there was no pre-arm switch before use that as a security measure too) you should be forced to move the arm-switch to disarm before you can arm the quad.

DanNixon, 2017-08-22:
Can you provide the output of `version` and `diff all` from the CLI from the version 3.2 setup.

Earlier builds of 3.2 will not have had the additional protection I refer to.
I'm still very unclear on how you have your modes or failsafe setup.

mStrangers, 2017-08-22:
it's the diff of my 3.1.7 on my 3 inch all other craft is at my workshop
`# Betaflight / PIKOBLX 3.1.7 Apr  3 2017 / 22:16:26 (e1c4b5c)

feature -VBAT
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature AIRMODE
map TAER1234
serial 0 8192 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 0 1300 1700
aux 1 28 0 1300 1700
aux 2 2 1 1300 1700
aux 3 27 2 1700 2100
aux 4 13 2 1300 1700
set small_angle = 180
set serialrx_provider = SBUS
set current_meter_type = VIRTUAL
set failsafe_kill_switch = ON
set acc_hardware = NONE
set pid_process_denom = 3
profile 0

rateprofile 0
rateprofile 0

set rc_expo = 14
set rc_yaw_expo = 14
set thr_expo = 10
set roll_srate = 77
set pitch_srate = 78
set yaw_srate = 88

# `

mStrangers, 2017-08-22:
the 3.2 version i use is the rc3 downloaded from the configurator

mStrangers, 2017-08-22:
i arm the quad with a switch prop spin.
i activate failsafe killswitch ( arm switch always set to arm): prop stop spining and buzzer start is sound, failsafe stage 2.
in 3.1.7 if i toggle the failsafe killswitch to off ,nothing append except some sound from buzzer , i have to move the arm switch to disarm then to arm for the quad arm and prop spin.
in 3.2 RC3 if i toggle the failsafe killswitch to off ,the quad arm and prop spin.(no prearm switch)

DanNixon, 2017-08-22:
I can't see any reason for this not to be working in RC3, I'll look into it more in the morning.

For now can you try a nightly build and see if you still have issues.

mStrangers, 2017-08-22:
i can test it only tomorrow ,there's no nightly for the picoblx

DanNixon, 2017-08-22:
`betaflight_3.2.0_FF_PIKOBLX-107-326db13f2.hex` should be right (that is from the last successful job).

mStrangers, 2017-08-22:
ops yes i searched a P not FF
on the piko it work like 3.1.7 with 3.2 RC3 and nightly (no problem on this board)
i'll test nightly tomorow with the omibus f4 pro corner that the problem appear today

mStrangers, 2017-08-22:
i've r tested with the piko in the exact same procedure that the problem appear with the omnibus and this bug reappear
if i plug lipo with failsafe killswitch activated and arm switch to arm when you toggle failsafe killswitch to off , quad do nothing in 3.1.7 , arm in 3.2 rc3 nightly didn't tested it didn't arm after reflash

mStrangers, 2017-08-23:
tested with lastest nightly #109 betaflight_3.2.0_OMNIBUSF4SD-109-7a6626071.hex
the bug are not present in this version only in 3.2 RC3
So i thing you can close

closed 2017-08-23 by martinbudden