betaflight/betaflight #3927: bf 3.2.0 - sdcard functionality broken

TheIsotopes, 2017-08-22:
since [jenkins build 1589](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1589/) is sdcard functionality broken an my spracingf3evo ... sdcard inserted and the fc hangs on initialisation ... sdcard removed and fc working normal ... imho #3907 is the troublemaker ... @jflyper pls can you see whats wrong

TheIsotopes, 2017-08-22:
imho it's only affected on SPI2 ... sdcard from f3evo runs on SPI2 and this #3928 also runs on SPI2 ... my revolt with onboardflash running on SPI3 has not this issue

jflyper, 2017-08-23:
@TheIsotopes 
The cases are aggregated in #3932.
Can you provide some information about the nature of your board?
Genuine, clone, product URL?

closed 2017-08-23 by jflyper

TheIsotopes, 2017-08-23:
@jflyper it's an genuine board mk2 from http://seriouslypro.com/spracingf3evo

richard-scott, 2017-08-23:
@jflyper Same board as me.