betaflight/betaflight #3929: SP F3 Evo doesn't work with firmware after #01873d3f8 was merged.

richard-scott, 2017-08-22:
I have an SP Racing F3 Evo board, that is working fine with build [#103](https://ci.betaflight.tech/job/Betaflight/103) from ci.betaflight.tech, but with build [#104](https://ci.betaflight.tech/job/Betaflight/104) (to [#107](https://ci.betaflight.tech/job/Betaflight/107) which is the latest available right now) it doesn't work fully. When connected to USB (after flashing) I get the initial booting chirps from my on-board buzzer, but not the multi beeps that says its ready.

When BFC attempts to talk to it the board times out after the initial comms:

[image]

Nothing I do restores functionality. I've tried re-downloading the .hex, flashing via dfu etc etc.
Something in [here](https://ci.betaflight.tech/job/Betaflight/104/changes) broke the firmware compatibility with this board. I'm getting a response from the Com Port, so I don't believe its a computer or drivers issue.

TheIsotopes, 2017-08-22:
do you have the sdcard inserted ... remove it and the fc boots normal ... imho same issue like this #3927

jflyper, 2017-08-23:
@richard-scott 
The cases are aggregated in #3932.
Can you provide some information about the nature of your board? Genuine, clone, product URL?

closed 2017-08-23 by jflyper