betaflight/betaflight #3932: Flash and SDCARD stopped working after #3907

jflyper, 2017-08-23:
#3907 fixed a bug that caused SPI2 and SPI3 buses at half the intended speed. The fix was universal, and it will consume half the CPU cycles for writing to flash and SDCARD when DMA is not used.

Unfortunately, there are some cases the fix broke flash and SDCARD to work.

I would like to aggregate such cases here.

jflyper, 2017-08-23:
SPRACINGF3EVO, SDCARD wedges FC, #3927, #3929.

jflyper, 2017-08-23:
XRACER_SPI, data flash not detected. #3928.

richard-scott, 2017-08-23:
I have a geniune SP F3 Evo.

When I flash it with #103, the board reboots, the buzzer beeps fast for about a second and then beeps about 6 times times to say USB connected.

However, when I flash #104 or above, the buzzer beeps the inital boot beeps, but then nothing. The serial port responds initially as BFC can get the board details, but when BFC attempts to load the current config it times out.

I did a full chip rease as shorting out the boot pins is the only way to re-flash back to #103 as the BFC's automatic reboot into DFU mode doesn't work in #104.

I have a Jenkins server at home that builds the latest codebase from GitHub, so I'm more than happy to test anything required.

TheIsotopes, 2017-08-23:
@jflyper imho i found the problem
in #3907 you halves the divisor and in the `target.h` from sprf3evo is also a divisor defined `#define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER     2` ... now this makes a quarter, or ;-) ... im now set `#define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER     1` and my evo boots again with sdcard inserted

richard-scott, 2017-08-23:
Funny. I just came back to say my SPF3Evo had an SD card inserted during the problem with booting. I was going to say that I will remove it later tonight when I get home to see if that helps... looks like you've found the issue tho.

jflyper, 2017-08-23:
@TheIsotopes No, that is not the problem. And if you have set `SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER` to `1`, then your clock is even faster.

I have a fix in #3936.

TheIsotopes, 2017-08-23:
@jflyper #3936 tested now and it works ... nice ;-)

jflyper, 2017-08-23:
@TheIsotopes Thanks for testing! 👍

closed 2017-08-23 by martinbudden

richard-scott, 2017-08-23:
@jflyper Things are working for me now too (not that you expected otherwise given I've got the same board) 👍

jflyper, 2017-08-23:
@richard-scott and @here Thanks for testing, and apologies for my oversight.

mikeller, 2017-08-25:
@here actually works?

richard-scott, 2017-08-25:
@jflyper no problem, that's what testing is all about :-)

jflyper, 2017-08-26:
@mikeller Do you still have a problem?

mikeller, 2017-08-26:
@jflyper: Sorry for being confusing. My question was about the `@here` command working in GitHub. :-)

jflyper, 2017-08-27:
Ah... I think you don't need @here as default behavior is to notify everyone listed as participants.

mikeller, 2017-08-27:
Yeah, I guess so. ;-)